framps/raspiBackup #1: Support raspi3 if no SD card is used any more

framps, 2016-12-11:
As of now raspiBackup requires a SD card on /dev/mmcblk0. raspi3 allows to run without SD card. Users with raspi3 cannot use raspiBackup.

framps, 2016-12-11:
The normal backup mode is already implemented and will be available in next release. Support for partition oriented mode is still missing but worked on already.

timtimsson, 2017-01-14:
Hello,
i use also a Raspi3 with only a USB-stick without sdcard.
If i start raspiBackup.sh ,the script don't found the sd and break...
(0.6.1.3b)

```
RBK0017E: Keine SD Karte /dev/mmcblk0 gefunden
RBK0005E: Backup fehlerhaft beendet: Miscellaneous error (RC: 114)

```

framps, 2017-01-14:
@timtimsson: There are still issues with the beta version and there will be a beta2 published. Please add a comment on https://www.linux-tips-and-tricks.de/en/raspberry/486-raspibackup-raspberry3-support and provide your eMail address (will be hidden and visible  by me only)

framps, 2017-04-09:
Yesterday version 0.6.2 was published which supports Raspi3 running in USB boot mode

closed 2017-04-09 by framps