framps/raspiBackup #4: Multiple processes are started

ReneHezser, 2017-02-05:
When I start a backup via cron or manually, it runns smoothly for quite some time.

> --- RBK0009I: orangepipc: raspiBackup.sh V0.6.1.3b um Sun Feb  5 20:48:25 CET 2017 gestartet
--- RBK0128I: Logdatei ist /mnt/BackupNas/orangepipc/orangepipc-tgz-backup-20170205-204823/orangepipc-backup.log
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt
--- RBK0008I: Services werden gestoppt: '/etc/init.d/samba stop'
--- RBK0081I: Backup vom Typ tgz wird in /mnt/BackupNas/orangepipc/orangepipc-tgz-backup-20170205-204823 erstellt
--- RBK0085I: Backuperstellung vom Typ tgz läuft. Bitte etwas Geduld
--- RBK0036I: Partitionslayout wird gesichert
--- RBK0044I: Backup der Bootpartition wird in /mnt/BackupNas/orangepipc/orangepipc-tgz-backup-20170205-204823/orangepipc-backup.img erstellt
--- RBK0044I: Backup des Partitionlayouts wird in /mnt/BackupNas/orangepipc/orangepipc-tgz-backup-20170205-204823/orangepipc-backup.sfdisk erstellt
--- RBK0046I: Backup des Masterbootrecords wird in /mnt/BackupNas/orangepipc/orangepipc-tgz-backup-20170205-204823/orangepipc-backup.mbr erstellt

After a while, a new backup is started every minute.
This is an extract from ps -ef:

> root     16412     1  0 22:02 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     16528 16412  0 22:02 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     16529 16412  0 22:02 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     16530 16528  0 22:02 ?        00:00:00 tee -a /home//orangepipc-raspiBackup.log
root     16532 16529  0 22:02 ?        00:00:00 tee -a /home//orangepipc-raspiBackup.log
root     16752 16412  1 22:02 ?        00:00:04 dd if=/dev/mmcblk0p1 of=/mnt/BackupNas/orangepipc/orangepipc-tgz-backup-20170205-220202/orangepipc-backup.img bs=1M
root     16815     1  0 22:03 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     16915 16815  0 22:03 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     16916 16815  0 22:03 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     16917 16915  0 22:03 ?        00:00:00 tee -a /home//orangepipc-raspiBackup.log
root     16918 16916  0 22:03 ?        00:00:00 tee -a /home//orangepipc-raspiBackup.log
root     17128 16815  0 22:03 ?        00:00:03 dd if=/dev/mmcblk0p1 of=/mnt/BackupNas/orangepipc/orangepipc-tgz-backup-20170205-220302/orangepipc-backup.img bs=1M
root     17207     1  0 22:04 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     17323 17207  0 22:04 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     17324 17207  0 22:04 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     17325 17323  0 22:04 ?        00:00:00 tee -a /home//orangepipc-raspiBackup.log
root     17327 17324  0 22:04 ?        00:00:00 tee -a /home//orangepipc-raspiBackup.log
root     17547 17207  0 22:04 ?        00:00:02 dd if=/dev/mmcblk0p1 of=/mnt/BackupNas/orangepipc/orangepipc-tgz-backup-20170205-220401/orangepipc-backup.img bs=1M
root     17622     1  0 22:05 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     17713 17622  0 22:05 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     17714 17713  0 22:05 ?        00:00:00 tee -a /home//orangepipc-raspiBackup.log
root     17715 17622  0 22:05 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     17717 17715  0 22:05 ?        00:00:00 tee -a /home//orangepipc-raspiBackup.log
root     17937 17622  0 22:05 ?        00:00:02 dd if=/dev/mmcblk0p1 of=/mnt/BackupNas/orangepipc/orangepipc-tgz-backup-20170205-220501/orangepipc-backup.img bs=1M
root     18026     1  0 22:06 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     18168 18026  0 22:06 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas
root     18169 18026  0 22:06 ?        00:00:00 /bin/bash /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas

This is the command executed: `/usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas &> /dev/null`

framps, 2017-02-06:
A similar issue was already reported some time ago by a user of raspiBackup. Actually raspiBackup has no means to guarantee it's executed only once. I will add this check in the next version now.
The root cause was a misconfigured crontab entry which kicked off raspiBackup every 5 minutes. 

`/usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas &> /dev/null` seems to be part of your crontab. Please check this entry.

ReneHezser, 2017-02-06:
`* 22 * * 0 /usr/local/bin/raspiBackup.sh -t tar -e email@tld.de -k 30 -z -n /mnt/BackupNas &> /dev/null`
You are right! How could I miss that one :-(
Thank you.

closed 2017-02-06 by ReneHezser

framps, 2017-02-06:
Great you fixed it now :-) Just added code in the next version 0.6.1.4 which will write an error message if raspiBackup is started when it's already up and running