framps/raspiBackup #5: Possible to see the Status running on commandline

skyerjoe, 2017-03-05:
Hello Framp 
 

Is it possible to integrate an program like pv to see the  Status from the raspibackup when your are starting the program from the commandline  ?

for ex. :  dd if=/dev/sdX **| pv |** dd of=/pfad/zur/datei

btw. i tried to create an message on your Website but i cannot see any Picture for captcha verification ..

regards John

framps, 2017-03-05:
Hello John,

I have no idea why you don't see any captcha verification. It works for me and others :-)  

Feel free to send me an eMail you will find on the contact page.

Anyhow: It's no problem to add an additional option which cause raspiBackup to use pv. But raspiBackup is supposed to run in the background via crontab.  What's your usecase to have a progressbar? The only one I can think of is when the restore is executed. Do you have this usecase in mind?

Cu framp

framps, 2017-04-01:
@skyerjoe Just thinking about this feature request. I agree it's a useful feature when a backup is restored. Sounds to me you also would like to have the progress back when creating a backup. What's the rational for this?

closed 2017-08-03 by framps

framps, 2017-08-24:
Other people asked for this feature also. I'm now implementing it for backup and restore. This feature will be available with the next version.