framps/raspiBackup #7: Permissions after restore

lorenzogm, 2017-05-12:
Hi @framps 

I have used your code to backup and restore a raspbian created with NOOBS. It was a dd backup and it has been restore with `sudo ./raspiBackup.sh -Z -d DEVICE BACKUP`

It looks it's working but the permissions of the filesystem are wrong (root is not the owner anymore):

[image]

And I get errors like `sudo: /usr/bin/sudo must be owned by uid 0 and have the setuid bit set`.

Any idea how can I fix it?

Thank you!

framps, 2017-05-12:
Hi Lorenzo,

that's very nasty and shouldn't happen :-( . Are you sure the image you saved haven't had the incorrect ownership already? I frankly don't see any other explanation how this could have happened.

> Any idea how can I fix it?

1) If you have multiple backup versions (I keep 12 weekly backups) just got back the versions until you find one which doesn't have the owner issue. Just mount the dd image and check whether the ownership is correct. I

2) You can correct the restored image and change the owner manually. Start another image on your system, mount the incorrect image and correct the owners of all files. But that's very time consuming and I think it's better to create a fresh new image and restore all the important data from the restored image. 

Hope this helps.

framps, 2017-05-16:
@lorenzogm Do you have any updates ?

lorenzogm, 2017-05-18:
Hi @framps, it's working now. I fixed the permissions in the source and it perfectly working now. Thank you!

closed 2017-05-18 by lorenzogm