framps/raspiBackup #8: Unable to boot a backup restored backup when using Jessie and an external root filesystem

framps, 2017-06-05:
Root cause is that Jessie uses partuuid instead of partition names now. Therefore a dumb restore of the backup when using another external usb partition doesn't work any more and some postprocessing of /boot/cmdline.txt and /etc/fstab is required ahen the backup was restored.

tnmils, 2017-06-06:
Hi,
i have the same problem i think, i made a backup (rsync) of rootfs which is on external hdd and tried to restore on a single SD card.  i checked my fstab  (of restored SD card)  which is

`proc            /proc           proc    defaults          0       0
/dev/mmcblk0p1  /boot           vfat    defaults          0       2
/dev/mmcblk0p2  /               ext4    defaults,noatime  0       1
`

and changed back my cmdline.txt to have root located again on mmcblk0p2 instead of /dev/sdaX

`wc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
`
but no success. 
message from console also gives an error and error codes code 23 and 95 

> 
sudo raspiBackup.sh -m detailed
--- RBK0009I: myhost: raspiBackup.sh V0.6.2 (aad4efd) started at Tue Jun  6 08:52:14 CET 2017
--- RBK0128I: Using logfile /backup/myhost/myhost-rsync-backup-20170606-085213/myhost-backup.log
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf
--- RBK0151I: Using backuppath /backup
--- RBK0008I: Stopping services: 'service nginx stop & service php5-fpm stop & service postfix stop & service dovecot stop & service cron stop & service mysql stop'
--- RBK0081I: Creating backup of type rsync in /backup/myhost/myhost-rsync-backup-20170606-085213
--- RBK0036I: Saving partition layout
--- RBK0044I: Creating backup of boot partition in /backup/myhost/myhost-rsync-backup-20170606-085213/myhost-backup.img
--- RBK0045I: Creating backup of partition layout in /backup/myhost/myhost-rsync-backup-20170606-085213/myhost-backup.sfdisk
--- RBK0046I: Creating backup of master boot record in /backup/myhost/myhost-rsync-backup-20170606-085213/myhost-backup.mbr
--- RBK0133I: Using directory /backup/myhost/myhost-rsync-backup-20170605-095910 for hardlinks
--- RBK0158I: Creating native rsync backup "/backup/myhost/myhost-rsync-backup-20170606-085213"
--- RBK0085I: Backup of type rsync started. Please be patient
??? RBK0024E: Backup tool rsync received an error. 
rsync: rsync_xal_set: lsetxattr(""/backup/myhost/myhost-rsync-backup-20170606-085213/usr/bin/systemd-detect-virt"","security.capability") failed: Operation not supported (95)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1183) [sender=3.1.1]
!!! RBK0049W: Some files were changed or vanished during backup. RC 23 - ignoring change
--- RBK0007I: Starting services: 'service nginx start & service php5-fpm start & service postfix start & service dovecot start & service cron start'
--- RBK0010I: myhost: raspiBackup.sh V0.6.2 (aad4efd) stopped at Tue Jun  6 09:22:08 CET 2017
--- RBK0017I: Backup finished successfully

framps, 2017-06-06:
Hi tnmils,
please note: You have to update the root path in two places:
1) /boot/cmdline.txt
2) /etc/fstab

Re your xattr issue: That's not related to this issue. It's a known issue with extended attributes and nfs. rsync cannot save the extendend attributes because nfs does not support extended attributes even the used filesystem is ext4 which supports extended attributes.

You have following options:
1) Remove the extended attributes from your system. Usually they are not that important.
2) Remove -X from DEFAULT_RSYNC_BACKUP_OPTIONS="-aHAXx" and don't save extended attributes
3) Save the backup on a direct attached USB disk with an ext4 partition
4) Use tar backup

Hope this helps :-)

framps, 2017-06-13:
Issue fixed. raspiBackup now has a postprocessing step which updates the PARTUUIDs. Fix will be available with next raspiBackup version. See https://www.linux-tips-and-tricks.de/en/versionhistory for details.

closed 2017-06-13 by framps