framps/raspiBackup #14: Not possible to turn off -z option in command line if it's turned on in config

framps, 2017-10-08:


framps, 2017-10-08:
Fixed in https://github.com/framps/raspiBackup/commit/58febd1e8a6e1c4888ab7f1a459431680ee3b3a0

closed 2017-10-08 by framps

christianTF, 2017-10-08:
Hi @framps ,
thanks for the change, but to toggle is, I think, not a good solution.
Working with command line alone will lead to an unpredictible result if -z toggles.

I'm working on a sort of web interface for raspiBackup (it's a plugin for https://github.com/mschlenstedt/Loxberry) - it does not run standalone. 

If I call raspiBackup.sh by command line (cron), I first need to parse the default config to know if -z enables or disables compression. A dedicated switch would be preferable, and -z (and the default config) should be ignored on rsync backup-type.

Cool backup tool. You'll be credited in my interface when it's done.

cu, Christian

framps, 2017-10-08:
Hello @christianTF ,

I thought you're another raspiBackup user and didn't know you have a special use case :-)

It makes sense to be able to toggle options via commandline. That's why I immediately fixed the logic.  Actually it would make sense to change the option handling in raspiBackup to be able to set or unset an option. Maybe I'll work on this at some point in time. 

There is another undocumented way to **override** the config options: raspiBackup parses 3 config files in following sequence:

1. /usr/local/etc/raspiBackup.conf
2. /home/$USER/.raspiBackup.conf
3. $PWD/.raspiBackup.conf

If you just create either (2) or (3) and insert 
`DEFAULT_ZIP_BACKUP=0`
the parameter in (1) will be overridden and you're fine.

Let me know if this is a valid option for you.  We'll find a solution for you. Don't hesitate to create new issues if there is any missing feature for you in raspiBackup. But then it will be helpful to explain your use case in more detail ;-) 

Cu framp

christianTF, 2017-10-08:
Hi framp,

thanks for this info, I will check that. I think this is my solution :-)

Cu, Christian