framps/raspiBackup #16: Feature Request: Backup /etc/fstab as extra file

christianTF, 2017-10-22:
Hello framp!

For simplification of a restore, I'd like to request that every backup type should copy the /etc/fstab file as an extra file into the backup directory as plain text file.

I've noticed that especially "Windows" guys (that also involves me) have troubles to recover the mounts to re-access the backup volume in case of a restore (or at least it is time consuming to figure out the settings).
With a plain copy of the fstab it is a copy/paste job to re-enable the backup mount to further process the restore operation.

Regards, Christian

framps, 2017-10-22:
Hello Christian,

it shouldn't be a big deal for me to implement your request. But first I want to get the rational for the request. Maybe I have another solution handy :wink: 

Looks like you want to provide some information to the Windows user so he knows how he can mount the partitions restored on the SD card for further editing of some restored files. Why should a user want to do this? A restore should be an exact copy of the original system and not modified at all. Maybe you can give me an example.

christianTF, 2017-10-22:
Hi,

As I have understood, a restore under Raspbian looks like this:
1. Setup Raspbian on another SD
2. A second SD card must be placed in a SD card reader connected to the Raspbian USB
**3. Re-mount the backup volume**
4. Call the restore with source and destination

Number 1 can be managed easily as this is a standard routine. 
Number 2 might need some try&error to find out the right /dev/sdx (but the possibilities are limited)
Number 3 in fact will be some efford if disks are mounted by UUID, or shares have options and credentials
Number 4 can be automated if we have finished number 2 and remounted /backup from number 3.

Therefore, the most difficult part is remounting the backup volume, as everyone will have forgotten how it was configured times ago.

I just have finished the first part of my WebUI for LoxBerry, to ad hoc run and schedule backups by webinterface.
The second part will be some supporting stuff for a restore of TGZ and RSYNC backups in the web interface. 

BTW, it is meant as feature request for future versions. Possibly you get out some other ideas how to more automate restores. I've read about the automatic image restorer script, but had not found more documentation about it. That are great news!
Currently, I can copy the fstab in my UI too, so it is not an urgent wish.

Regards,
Christian

http://www.loxwiki.eu/display/LOXBERRY/LoxBerry+Backup

framps, 2017-10-22:
Thank you very much for the update. I now see the usecase: A Windows user who primed the Raspberry with LoxBerry, just runs it and has no Linux experience should get detailed instructions how to restore the backup on his Raspberry. That's very useful because in general a user is quite nervous when the SD card broke and he has to restore the backup. I love this idea.

Do you plan to parse /etc/fstab to generate a script or to provide detailed steps how to mount the backup partition? Actually this assumes the backup partition is mounted in /etc/fstab which may not be true in all environments. But if you document this requirement appropriately it should work :smile: 

I followed your link to LoxBerryBackup and it looks pretty good. :+1: 

Frankly I don't like to add the copy function of /etc/fstab into raspiBackup which will pollute raspiBackup with special requirement code. But I think it's a nice idea to add another extensionpoint to raspiBackup which is called just before the backup with dd, tar or rsync is started. The target backup directory will be available as a bash variable and can be used to do everything you want. You only have to provide a small bash script which copies /etc/fstab to the backup directory :smiley: Other raspiBackup users can do other stuff in this extension point. For details about extension points see [here](https://www.linux-tips-and-tricks.de/en/13-raspberry/443-raspibackup-extensions)

christianTF, 2017-10-22:
The plugin/extension is a good idea, as I cannot know the backup path outside.

As "my" LoxBerry user usually sits in front of a Windows PC and web browser, parsing the fstab from users upload is really a good idea! (assuming the user can access the file from a share, e.g. NAS).

My restore concept for LoxBerry image would be:
1. Write the Loxberry image to a new SD and boot
2. Insert the destination SD to a card reader and plugin to Raspberry
3. Re-Install the backup plugin
4. Upload fstab from the backup device and my plugin will try to restore and mount (or manually restore the mountpoint)
5. Select source backup
6. Select destination /dev/sdx
7. Go!

Thanks for your inspiration!

framps, 2017-10-22:
Great you accept my proposal. I reopened the issue and will close it when I added the additional extension point in raspiBackup.

framps, 2017-10-25:
@christianTF 
https://github.com/framps/raspiBackup/commit/cd24e8fd68c7d581308b48df6dfa6b7c90553f5f inlcudes a modified version of raspiBackup which supports the discussed ready extension. In addition a sample read extension which copies /etc/fstab into the backup dir is included. Add
```
DEFAULT_EXTENSIONS="fstab"
```
in the config file to enable the sample extension.

Please test whether this works for you.

christianTF, 2017-10-26:
@framps 
Thanks for your work!
I first was wondering that it was not working, but later saw in your commit that you have updated the main program as well to support the _ready postfix.
It works now, and have updated raspiBackup to your dev branch in my Loxberry plugin.

This solution fulfills my expectation! Thanks

framps, 2017-10-26:
Great. Thank you very much for testing. I'll close this issue when the change is available in the next release.

framps, 2017-11-05:
Fixed in v0.6.3.1

closed 2017-11-05 by framps