framps/raspiBackup #23: raspiBackupRestore2Image.sh fails RC112

Taifun2300, 2017-10-26:
From the log:

20171026-162244: DBG              << calcSumSizeFromSFDISK 64013265920
20171026-162244: DBG              -- sourceSDSize: 64013265920 - targetSDSize: 64012419072
20171026-162244: DBG              -- sourceBlockSize: 124932096 - adjusted targetBlockSize: 124930442
20171026-162244: MSG              !!! RBK0004W: Zweite Partition wird von 59.57 GiB auf 59.57 GiB angepasst.
20171026-162244: DBG                 >> resizeRootFS
20171026-162244: DBG                 -- RESTORE_DEVICE: /dev/loop0
20171026-162244: DBG                 -- ROOT_PARTITION: /dev/loop0p2
20171026-162244: DBG                 -- partitionLayout of /dev/loop0
20171026-162244: DBG                 -- Disk /dev/loop0: 59,6 GiB, 64012419072 bytes, 125024256 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x19aa3ab1

Device       Boot Start       End   Sectors  Size Id Type
/dev/loop0p1       8192     93813     85622 41,8M  c W95 FAT32 (LBA)
/dev/loop0p2      94208 125026303 124932096 59,6G 83 Linux
20171026-162245: DBG                 -- PartitionStart:
20171026-162245: MSG                 ??? RBK0111E: Fehler  beim Erstellen der Partitionen.RC
.
20171026-162245: DBG                    >> exitError 112
20171026-162245: DBG                 << exitError 112
20171026-162245: DBG                    >> cleanup
20171026-162245: DBG                       >> cleanupRestore
20171026-162245: DBG                       -- Got trap EXIT
20171026-162245: DBG                       -- rc: 112
20171026-162245: DBG                          >> isMounted /tmp/raspiBackup
20171026-162245: DBG                          -- /dev/root
20171026-162245: DBG                       << isMounted 1
20171026-162245: DBG                       -- Deleting dir /tmp/raspiBackup
20171026-162245: MSG                       ??? RBK0077E: Restore wurde fehlerhaft mit RC 112 beendet. Siehe vorhergehende Fehlermeldungen.
20171026-162245: DBG                    << cleanupRestore - exit with 112
20171026-162245: DBG                       >> cleanupTempFiles
20171026-162245: DBG                       -- Removing mailfile /root/script/raspiBackup.maillog
20171026-162245: DBG                    << cleanupTempFiles
20171026-162245: DBG                    -- Terminate now with rc 112

framps, 2017-10-26:
Unfortunately that's not enough information. I need the complete debug log for further analysis.

Taifun2300, 2017-10-26:
Log file 
[raspiBackup.log](https://github.com/framps/raspiBackup/files/1419842/raspiBackup.log)

framps, 2017-10-28:
You try to restore a raspiBackup into a loop device which fails. Actually I created a [script](https://github.com/framps/raspiBackup/blob/master/raspiBackupRestore2Image.sh)  a couple of weeks ago which exactly does this and calles pishrink afterwards. I also had issues to restore directly into the loop device and was not able to figure out the root cause. Looks like it's somehow related to the kernel and refresh of a modified partition table. The workaround I use is to prime the loop device with the partition info of the backup first and then start the restore.  Just check the script raspiBackupRestore2Image.sh in this repo.

framps, 2017-11-01:
@Taifun2300 Any news on this issue from your side?

Taifun2300, 2017-11-02:
I just downloaded the latest version, but got unfortunarly the same result.

framps, 2017-11-02:
There is a beta version of raspiBackup for v0.6.3.1 available with some changes related to your issue. Please try [the beta version](https://raw.githubusercontent.com/framps/raspiBackup/beta/raspiBackup.sh) of raspiBackup

Taifun2300, 2017-11-03:
No it works :-) 
Danke!

framps, 2017-11-05:
Fixed in v0.6.3.1

closed 2017-11-05 by framps