framps/raspiBackup #25: Language C not supported.

immae1, 2017-11-09:
Hi i get the error Message ??? RBK0096E: Language C not supported. 
How could i fix this ? 
Many thanks.

framps, 2017-11-09:
Hi immae1,
actually this should not happen :-(. raspiBackup should fall back to English if your local language is not supported. Looks like this is a bug I have to fix.

You can fix this:
1. Add DEFAULT_LANGUAGE="de" or DEFAULT_LANGUAGE="en"
2. Use option -G and pass de or en

Please do me a favor and tell me the result of following command executed on your Raspberry:
`echo $LANG`

framps, 2017-11-09:
I just fixed the issue and updated raspiBackup.sh v0.6.3.1. Just [download raspiBackup](https://www.linux-tips-and-tricks.de/raspiBackup.sh) again to grab the new version :-)

framps, 2017-11-12:
@immae1 Please verify my fix.

immae1, 2017-11-13:
@framps yes now it works. i used curl like this for the new version: 
curl -s -L -O https://www.linux-tips-and-tricks.de/raspiBackup.sh

framps, 2017-11-14:
Great. Thank you very much for verifying.

closed 2017-11-14 by framps