framps/raspiBackup #27: RaspiBackup cannot run withour perl

wemsras, 2017-11-24:
Please check if perl installed. If perl is not installed, then you get empty lines with no information. There is no error message. You only see blank lines.  I think of testing for perl in the begining of the script like the check fpr BASH.

framps, 2017-11-24:
Good catch :+1: I will either add a check for perl of even remove the perl dependency.

framps, 2017-11-24:
raspiBackup will not depend on perl any more in next release. I cherry-picked the fix in https://github.com/framps/raspiBackup/tree/detect_mountpoint. Please verify.

wemsras, 2017-11-24:
Yeah. Good job.
i have removed perl an after that i run the raspiBackup.sh (Hotfix Version) without problems. 
Thank you.
[rbwoperl.txt](https://github.com/framps/raspiBackup/files/1502485/rbwoperl.txt)

framps, 2017-11-24:
Great. :+1: Will merge the fix into master for next version.

framps, 2018-03-10:
https://github.com/framps/raspiBackup/releases/tag/0.6.3.2

closed 2018-03-10 by framps