framps/raspiBackup #28: no backup on systems with only "one" partition

wemsras, 2017-11-24:
on a beaglebone black you can install your linux on "one" partiton. Specialy to the on-board EMMC storage.  You get the error ??? RBK0142E: Unable to detect boot device. The /boot is on the / file system and there is no seperate mountpoint for /boot. This beaglebone is running Parabola Linux.
[onefilesystem.txt](https://github.com/framps/raspiBackup/files/1501598/onefilesystem.txt)

framps, 2017-11-24:
Boot device detection assumes /dev/mmvblk0 as of now.

I just created a [test branch](https://github.com/framps/raspiBackup/tree/detect_mountpoint) for you which has dynamic boot device detection code added. Please try this version.

wemsras, 2017-11-24:
Sorry ;-)
It seems to be the same error with the new version.
[root@parabb bin]# ./raspiBackup.sh
--- RBK0009I: parabb: raspiBackup.sh V0.6.3.1-hotfix (b1e7cc4) started at Fri Nov 24 20:39:03 CET 2017.
!!! RBK0173W: =========> NOTE  <========= 
!!! RBK0173W: This is a temporary hotfix and should not be used in production. 
!!! RBK0173W: =========> NOTE <=========
??? RBK0142E: Unable to detect boot device.
--- RBK0105I: Deleting new backup directory /backup/parabb/parabb-tar-backup-20171124-203903.
??? RBK0005E: Backup failed. Check previous error messages for details.

and from /var/log/raspiBackup/parabb.log
20171124-203904: MSG           ??? RBK0142E: Unable to detect boot device.
Invocation parms: ''

framps, 2017-11-24:
:-( Could you please use option `-l debug` and provide the generated debug log? Make sure there is no sensitive information in the log.

wemsras, 2017-11-24:
Wow. Thanks for your fast reaction.
For me it looks like that the problem is that /boot is not a mountpoint, because of one filesystem /. The last lines of the attached log file shows something about that.
[parabb.log](https://github.com/framps/raspiBackup/files/1502463/parabb.log)

framps, 2017-11-24:
You are right. /boot is not a mount point on your system. raspiBackup requires /boot and / to be a mount point (that's standard on raspbian). If you want to use raspiBackup on your system the only backuptype you can use is dd. Let me know if you want to use dd instead of tar and I create another fix so you can use dd on your system.

wemsras, 2017-11-24:
On all machines i use tar for backup. For now i will check to change my config to /boot and /.
I try that out and come back asap.

wemsras, 2017-11-25:
At this time i was not able to change my BB black to boot from a two partition sd card. I've formated und partitioned the card several times without success. So i will analyze this at a later date. For now i close this issue. Many thanks for your "dd" offer.

closed 2017-11-25 by wemsras