framps/raspiBackup #29: always dd used for boot partition

jab42, 2017-12-07:
dd includes the risk of bad filesystem and corrupt image contents.
tar would avoid this.

framps, 2017-12-07:
Nice catch :+1: . Will enhance raspiBackup to save the root partition via tar and check for backblocks during restore.

framps, 2017-12-18:
Next version of raspiBackup will have the new option -B which forces raspiBackup to create a tar backup of the boot partition instead of a dd backup.

framps, 2017-12-18:
Branch [nextVersion](https://github.com/framps/raspiBackup/tree/nextVersion) now has the -B option to create a tar boot backup and the -C option to check for bad blocks when formating the restore SD card.

@jab42 : Please grab the new version and test :-)

framps, 2018-01-04:
@jab42 Please verify the new function requested.

framps, 2018-02-02:
@jab42 Please verify the new function now !!!. Current beta includes the function you requested.

framps, 2018-03-10:
https://github.com/framps/raspiBackup/releases/tag/0.6.3.2

closed 2018-03-10 by framps