framps/raspiBackup #31: Restore process hangs on security question (user confirmation)

phars, 2017-12-14:
I wanted to test a simple restore (.img file to device) but the script seems to be hanging when the user should confirm the restore:

```
--- RBK0009I: beemon: raspiBackup.sh V0.6.3.1 (590ae9c) started at Thu 14 Dec 12:48:07 CET 2017.
--- RBK0128I: Using logfile /raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0138I: Using bootbackup /backups/dd-backup-20171213-190402.img.
--- RBK0068I: Using bootpartition backup files starting with backup from directory /backups/dd-backup-20171213-190402.img.
!!! RBK0065W: Device /dev/sda will be repartitioned and all data will be lost.
```

I have to ctrl+c the script. I also noticed that the bash does not show any inputs anymore after stopping the script. Only `reset` revolves this issue. Seems to me that there could be some issues with reading from tty. Maybe I will find some time later to go deeper and find the root cause.

Or is there a known workaround?

phars, 2017-12-14:
Ok, when I connect via SSH with the gnome-terminal it works fine. I see the "Are you sure? y/N" question, can confirm and the script starts the restore. So there must be an issue with putty I guess (the error mentioned above is reproducible each time when I connect from a windows machine with putty).

framps, 2017-12-14:
I have no idea  what's wrong. I have to have a Windows Handy in order to reproduce the issue. Will try to find one

framps, 2018-01-02:
If this is still an issue please provide more details how I can reproduce the issue.

phars, 2018-01-02:
Sorry, didn't have the time to investigate further. For me it's not that critical since I do not use my windows machine very often. Thanks anyway!

Jakst3r, 2020-01-05:
I'm putting this here for anyone else who has this issue:

**Basically:**
Format the SD card you are restoring to.

**More Detail:**
I got this issue when trying to restore a backup to a smaller SD card to the original SD card the backup was taken from.
That SD card was previously used for a boot SD card in the RPI.

Tried again with a larger sd card (Didn't have another 32GB card) and it worked as expected.

Trying to restore to a smaller SD card that was previously used for RPI boot SD card:
```
sudo fdisk -l | egrep "^Disk /|^/dev"
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/mmcblk0: 29.7 GiB, 31914983424 bytes, 62333952 sectors
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 62333951 61801472 29.5G 83 Linux
Disk /dev/sda: 58.9 GiB, 63266881536 bytes, 123568128 sectors
Disk /dev/sdb: 14.9 GiB, 15931539456 bytes, 31116288 sectors
/dev/sdb1         8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/sdb2       532480 62333951 61801472 29.5G 83 Linux
```

```
sudo raspiBackup.sh -d /dev/sdb /backup/raspberrypi/raspberrypi-rsync-backup-20200104-125952/
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.4.3 (2d927a2) started at Sun  5 Jan 05:02:13 GMT 2020.
--- RBK0128I: Using logfile /home/pi/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0138I: Using bootbackup /backup/raspberrypi/raspberrypi-rsync-backup-20200104-125952.
--- RBK0068I: Using bootpartition backup files starting with /backup/raspberrypi/raspberrypi-rsync-backup-20200104-125952 from directory raspberrypi-backup.
!!! RBK0006W: Target /dev/sdb with 14.83 GiB is smaller than backup source with 29.72 GiB. root partition will be truncated accordingly. NOTE: Restore may fail if the root partition will become too small.
!!! RBK0065W: Device /dev/sdb will be repartitioned and all data will be lost.

IT JUST HABGS HERE
```

Formated the SD card to FAT

```
sudo fdisk -l | egrep "^Disk /|^/dev"
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
[... 18 lines cut ...]
!!! RBK0065W: Device /dev/sdb will be repartitioned and all data will be lost.
--- RBK0067I: Current partitions on /dev/sdb:
Number  Start   End      Size     Type     File system  Flags
 1      1.05MB  15931MB  15930MB  primary  fat32
!!! RBK0066W: Device /dev/sdb will be overwritten with the saved boot and root partition.
--- RBK0038I: Are you sure? y/N y
--- RBK0050I: Restoring backup from /backup/raspberrypi/raspberrypi-rsync-backup-20200104-125952.
--- RBK0051I: Restoring mbr from /backup/raspberrypi/raspberrypi-rsync-backup-20200104-125952/raspberrypi-backup.mbr to /dev/sdb.
--- RBK0052I: Creating partition(s) on /dev/sdb.
!!! RBK0004W: Adjusting second partition from 29.46 GiB to 14.58 GiB.
--- RBK0053I: Restoring first partition (boot partition) to /dev/sdb1.
--- RBK0054I: Formating second partition (root partition) /dev/sdb2.
--- RBK0055I: Restoring second partition (root partition) to /dev/sdb2.
--- RBK0184I: Rootpartition check started.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0076I: Restore finished successfully.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.4.3 (2d927a2) stopped at Sun  5 Jan 05:19:37 GMT 2020.

IT WORKED
```

Mac OS -> iTerm -> SSH - > RPI runing Buster

`/dev/sda` is a USB stick where I backup to / restore from
`/dev/sdba is the SD card I'm restoring to

framps, 2020-01-05:
@Jakst3r Thank you very much for the analysis and a workaround description. I will try to recreate the issue when restoring to a smaller SD card. If you still have the restore log handy please provide the log which may help me to identify the root cause for the issue :wink:

Jakst3r, 2020-01-05:
Not a problem :blush: Not anymore as it was overwritten on the successful restore after formating the card.

Il try and replicate the issue again and post the logs.

framps, 2020-01-11:
I just tried to reproduce your issue with the same code level you used. Unfortunately I failed :disappointed: 
```
sudo /home/pi/raspiBackup.sh -d /dev/sdc .
--- RBK0009I: raspberrypi-buster: raspiBackup.sh V0.6.4.3 (2d927a2) started at Sat 11 Jan 2020 07:40:11 PM CET.
--- RBK0128I: Using logfile /backup/raspberrypi-buster/raspberrypi-buster-rsync-backup-20200111-192728/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0138I: Using bootbackup /backup/raspberrypi-buster/raspberrypi-buster-rsync-backup-20200111-192728.
--- RBK0068I: Using bootpartition backup files starting with /backup/raspberrypi-buster/raspberrypi-buster-rsync-backup-20200111-192728 from directory raspberrypi-buster-backup.
!!! RBK0006W: Target /dev/sdc with 3.74 GiB is smaller than backup source with 7.50 GiB. root partition will be truncated accordingly. NOTE: Restore may fail if the root partition will become too small.
!!! RBK0065W: Device /dev/sdc will be repartitioned and all data will be lost.
--- RBK0067I: Current partitions on /dev/sdc:
Number  Start   End     Size    Type     File system  Flags
 1      1.05MB  269MB   268MB   primary  fat16        lba
 2      269MB   4025MB  3756MB  primary  ext4
!!! RBK0066W: Device /dev/sdc will be overwritten with the saved boot and root partition.
--- RBK0038I: Are you sure? y/N y
--- RBK0050I: Restoring backup from /backup/raspberrypi-buster/raspberrypi-buster-rsync-backup-20200111-192728.
--- RBK0051I: Restoring mbr from /backup/raspberrypi-buster/raspberrypi-buster-rsync-backup-20200111-192728/raspberrypi-buster-backup.mbr to /dev/sdc.
--- RBK0052I: Creating partition(s) on /dev/sdc.
!!! RBK0004W: Adjusting second partition from 7.24 GiB to 3.49 GiB.
--- RBK0053I: Restoring first partition (boot partition) to /dev/sdc1.
--- RBK0054I: Formating second partition (root partition) /dev/sdc2.
```
I used a different smaller size than you (4GB vs 16GB) but nevertheless the same codepath is used.

That's my environment:
```
sudo lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 279.5G  0 disk 
└─sda1        8:1    0 279.5G  0 part /backup
sdc           8:32   1   3.8G  0 disk 
├─sdc1        8:33   1   256M  0 part 
└─sdc2        8:34   1   3.5G  0 part 
mmcblk0     179:0    0   7.5G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0   7.3G  0 part /
```
The code which seems to hang is as follows:
```
function getPartitionTable() { # device

	logEntry "$1"
	logItem "$(IFS='' parted $1 unit MB p 2>>$LOG_FILE)"
	local table="$(IFS='' parted $1 unit MB p 2>>$LOG_FILE | sed -r '/^($|[MSDP])/d')"

	if [[ $(wc -l <<< "$table") < 2 ]]; then
		table=""
	fi
	echo "$table"

	logExit
}
```
Looks like parted hangs for some reasons...

I keep this issue open. If somebody faces the same issue please report the issue and we'll try to isolate the root cause.

closed 2020-03-14 by framps