framps/raspiBackup #33: rsync error

kmplngj, 2018-01-02:
Thanks for your great tool!

I have a problem with my installation.

I get this error message and do not know how to fix this.

`…
20180102-113322: DBG           -- mountpoint /boot: 179:1 mountpoint /: 179:2
20180102-113322: DBG           -- part: /dev/mmcblk0p1
20180102-113322: DBG           -- BOOT_DEVICE: mmcblk0
20180102-113322: DBG           -- BOOT_DEVICENAME: /dev/mmcblk0
20180102-113322: DBG              >> getPartitionPrefix: mmcblk0
20180102-113322: DBG           << getPartitionPrefix: mmcblk0p
20180102-113322: DBG           -- BOOT_PARTITION_PREFIX: mmcblk0p
20180102-113322: DBG        << inspect4Backup
20180102-113322: DBG           >> commonChecks
20180102-113322: DBG        << commonChecks
20180102-113322: DBG           >> getFsType: /media/nas-backup
20180102-113322: DBG        << getFsType:
20180102-113322: DBG        -- Found partitions on /dev/mmcblk0: 2
20180102-113322: DBG           >> isPathMounted: /media/nas-backup
20180102-113322: DBG           -- Path: /media/nas-backup
20180102-113322: DBG        << isPathMounted: 0
20180102-113322: DBG           >> supportsHardlinks: /media/nas-backup
20180102-113322: DBG           -- Links: 2
20180102-113322: DBG        << supportsHardlinks: 0
20180102-113322: DBG           >> supportsSymlinks: /media/nas-backup
20180102-113322: DBG           -- Links: 1
20180102-113322: DBG        << supportsSymlinks:
20180102-113322: DBG        -- rsync version: 3.1.2
20180102-113322: MSG        ??? RBK0019E: Option -a und -o nicht angegeben.
20180102-113322: DBG           >> exitError 107
20180102-113322: DBG        << exitError 107
20180102-113322: DBG           >> cleanup
20180102-113322: DBG              >> cleanupBackup
20180102-113322: DBG              -- Got trap EXIT
20180102-113322: DBG              -- rc: 107
…`

I hope you could help me.

Best,
Jan

kmplngj, 2018-01-02:
I just changed the backup method to tar and get the same error.

`20180102-114301: MSG        ??? RBK0019E: Option -a und -o nicht angegeben.`

framps, 2018-01-02:
Please read https://www.linux-tips-and-tricks.de/en/backup#parameters and https://www.linux-tips-and-tricks.de/en/faq#a18

closed 2018-01-02 by framps