framps/raspiBackup #36: Add option to define the sender eMailAddress

framps, 2018-02-13:
Right now the sender eMailAddress is hard coded. Make it configurable.

framps, 2018-06-15:
Config option DEFAULT_SENDER_EMAIL will be used. Default is root@$(hostname)

framps, 2018-07-01:
There is a beta available which includes a fix for this issue. Feel free to grab the [beta](https://github.com/framps/raspiBackup/tree/beta) and verify the fix.

DanSman, 2018-07-02:
Do we have to add "DEFAULT_SENDER_EMAIL" to our config?

# email to send completion status
DEFAULT_EMAIL="my@email.address"

# mailprogram
DEFAULT_MAIL_PROGRAM="ssmtp"

I'm running your beta but I get the same result like before. So, I have to change your script again at the mom.

framps, 2018-07-02:
Thank you very much for testing he beta. And yes, you have to define `DEFAULT_SENDER_EMAIL". Unfortunately this definition wasn't retrieved by raspiBackup. I just fixed it :smile: Please give it another try.

DanSman, 2018-07-02:
Hey

So, I went back to the stable version raspiBackup.0.6.3.2a and I deleted the beta and updated it again. There is still no change in the config you didn't fixed that, right?

I added "DEFAULT_SENDER_EMAIL" and everything is working fine now. It's changing the default sender email now as well. 

PS: The check of missing software is working fine here too. I'm using only rsync so that was fine anyway but I had missing software for don't know might be necessary for tar option or whatever. But it's working ;) So far so good. Thanks

framps, 2018-07-03:
> There is still no change in the config you didn't fixed that, right?

Ah ... I think I now get your point. The new config doesn't have the new variable included. That's a valid point. I don't update the config for a beta because then they are visible for people downloading the current non beta version which may cause trouble.I will add then when the beta will be pubished for everybody.

> I added "DEFAULT_SENDER_EMAIL" and everything is working fine now. It's changing the default sender email now as well.

Thank you for verifying th issue.

>PS: The check of missing software is working fine here too. I'm using only rsync so that was fine anyway but I had missing software for don't know might be necessary for tar option or whatever. But it's working ;) So far so good. Thanks

I thought about this. But then the code will be cluttered with various if/then/else statements. That's why I decided to test for everything which may be used.

framps, 2018-07-30:
Fixed in https://github.com/framps/raspiBackup/tree/v0.6.4

closed 2018-07-30 by framps