framps/raspiBackup #39: Add timestamps to logs being output by raspibackup

cgail914, 2018-03-09:
in essence 
1. add a timestamp to each log line generated
2. remove timestamping information inside the log itself

format of timestamp, I suggest a simple, easy to parse if needed
`date +'%m-%d-%Y %T'`

example instead of this 
_--- RBK0009I: jeedom-cugy: raspiBackup.sh V0.6.3.1 (590ae9c) started at Fri Mar  9 05:00:01 CET 2018.
--- RBK0128I: Using logfile /mnt/usblogs/jeedom-cugy/jeedom-cugy-rsync-backup-20180309-050001/jeedom-cugy-backup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0031I: Checking whether new version is available.
--- RBK0159I: Checking for beta version._

generate this
_03-09-2018 08:13:21 - RBK0009I: jeedom-cugy: raspiBackup.sh V0.6.3.1 (590ae9c) started
03-09-2018 08:13:21 - RBK0128I: Using logfile /mnt/usblogs/jeedom-cugy/jeedom-cugy-rsync-backup-20180309-050001/jeedom-cugy-backup.log.
03-09-2018 08:13:22 - RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
03-09-2018 08:13:22 - RBK0031I: Checking whether new version is available.
03-09-2018 08:13:22 - RBK0159I: Checking for beta version._

framps, 2018-03-09:
Hello @cgail914,

in general I think the most important times for raspiBackup users are start and endtime which is reported already. The debug log has timestamps in front of all messages using `date +%Y%m%d-%H%M%S`. 

I think most users will not be interested in leading timestamps in messages. In v0.6.3.1 I'm running out of options but next version v0.6.3.2 will have an extended options format and I can add option `--timestamps` which then will add leading timestamps in messages.

Does this sound OK for you?

cgail914, 2018-03-09:
Hello,
looks good to me. You're right it's definitely not a killer feature to add and on my side it is also more to align with a more standard way to display log lines, but if you can just add an option of timestamping, it'd be perfect. I admit I never tried the debug log - the script just works :-). Thanks again for the work, Best.

framps, 2018-03-10:
https://github.com/framps/raspiBackup/releases/tag/0.6.3.2

closed 2018-03-10 by framps