framps/raspiBackup #40: During version upgrade merge new config options into existing config file

framps, 2018-03-11:
Right now new options have to be added manually in the config files when a new version with new options is published. Would be great to have the new options being added during version update.

framps, 2018-10-03:
Current format of the config is unstructured (no JSON, XML or other structured format). Therefore it's not easy to merge configs. Therefore I'll put this issue in the backlog.

There is a workaround available: 
Install raspiBackup again with `curl -s -L -O https://www.linux-tips-and-tricks.de/raspiBackupInstall.sh && sudo bash raspiBackupInstall.sh `. The installer will save the old config file by appending the version number of raspiBackup to the filename. Then use a diff tool (I suggest meld on Linux) and compare and merge the old and new config files.

framps, 2020-04-07:
raspiBackup v0.6.5 will be published soon and has code freeze. There are new and changed options with 0.6.5 and the config file version changes from 0.1.3 to 0.1.4. In order to make the config merge as comfortable as possible create a bash script which merges the old an new config files by adding the 0.1.4 config statements prefixed with `#` to the corresponding statements in the old config file. That way a user can go through the config file and merge the config statements.

framps, 2020-04-07:
Created https://raw.githubusercontent.com/framps/raspiBackup/beta/helper/mergeConfigs.sh to help merging config files.

framps, 2020-04-08:
I just decided to add a new option `--mergeConfigs` into 0.6.5 beta. It will use a complete new code path and not influence other code thus it's added even the beta code is frozen already.

framps, 2020-04-08:
Unfortunately there is no way to get the merge enhancement added without to change base code.
In order not to break any existing code I created a [helper script](https://raw.githubusercontent.com/framps/raspiBackup/master/helper/mergeConfigs.sh) which merges the new config file with the current local config definitions as good as possible.

framps, 2020-04-09:
@pchristod On this [page](https://www.linux-tips-and-tricks.de/en/all-pages-about-raspibackup/566-raspibackup-how-to-update-configuration-when-upgrading-to-a-new-version/) I described how a configuration update can be done. Would you please check whether the instructions and the helper script help to make the upgrade convenient?

pchristod, 2020-04-09:
> @pchristod On this [page](https://www.linux-tips-and-tricks.de/en/all-pages-about-raspibackup/566-raspibackup-how-to-update-configuration-when-upgrading-to-a-new-version/) I described how a [...]

Can do that 👍 
Just to be clear, first step is to Upgrade to current 0.6.5 Beta, then following the steps described in the link you provided using the Helper script?

framps, 2020-04-09:
Great :+1: Thank you very much.

Well, the script downloads the current config from my website which is still the config for 0.6.4.3. So you will not get the config on 0.6.5-beta upgraded. There is no support to download the beta config. But as soon as the 0.6.5 beta will be published the script will download the 0.6.5 config and everything will work as expected.

I can change the script so it downloads the beta config and uses this one for upgrade instead of downloading the current config. Then you can upgrade to beta and merge the beta config.

But frankly - I'm just interested to get some feedback about the usability of the script and whether my doc how to merge is OK. So you also can just follow the description and generate an updated config and manually update it according my instructions. But finally you don't copy it in /usr/local/etc because it should look like the one you have active right now :wink: .

Just let me know whether you want to install the beta. I then will enable the script to download the beta config instead of the current config.

pchristod, 2020-04-09:
That's ok, good that I asked :)
I will do the dry run with my current stable version I have installed and the script just to get a feeling for it. Then I can upgrade to final 0.6.5 when that is out. Will let you know if everything worked and is understandable.

pchristod, 2020-04-09:
I just tested that. Everything worked fine. Explanations on the Page were pretty clear and easy to follow. I also find that this is a quite clever solution of the whole config mismatch (even if manual work is involved) as you can do that before upgrading which makes the upgrade a smooth experience. 

Only think I would probably add (just a suggestion) would be this info to the page that you can do this before the upgrade since the helpfer script fetches the most recents config file. That wasn't clear to me at a first glance, thats why I asked. 

Pretty good work, thank you!

framps, 2020-04-09:
Thank you for testing and your feedback :+1: 

> Only thing I would probably add (just a suggestion) would be this info to the page that you can do this before the upgrade since the helpfer script fetches the most recents config file. That wasn't clear to me at a first glance, thats why I asked.

Makes sense. I will add this information.

framps, 2020-04-23:
Upgrade process documented [here](https://www.linux-tips-and-tricks.de/en/configuration-update/). Helper script is in [0.6.5 beta branch](https://github.com/framps/raspiBackup/blob/beta/helper/mergeConfigs.sh) right now and will be merged into master when 0.6.5 will be published.

closed 2020-04-23 by framps

framps, 2020-04-26:
I decided to add the config update functionality in 0.6.5 and reopened https://github.com/framps/raspiBackup/issues/104