framps/raspiBackup #42: Backup failed.

caianhua1986, 2018-03-18:
--- RBK0085I: Backup of type tar started. Please be patient.

??? RBK0021E: Backupprogram for type tar failed with RC 2.
--- RBK0007I: Starting services: 'service nfs-common start %1%1 service cron start'.
--- RBK0026I: Saving logfile in /root/raspberrypi-raspiBackup.log.
--- RBK0043I: Removing incomplete backup in /backup/raspberrypi/raspberrypi-tar-backup-20180318-195037. This will take some time. Please be patient.
--- RBK0105I: Deleting new backup directory /backup/raspberrypi/raspberrypi-tar-backup-20180318-195037.
??? RBK0005E: Backup failed. Check previous error messages for details.

framps, 2018-03-18:
Linux tool tar which is used by raspiBackup fails with RC 2. See https://www.gnu.org/software/tar/manual/html_section/tar_19.html#Synopsis. Please use Option -v to get the detailed tar error messages.

framps, 2018-04-08:
@caianhua1986 Any update on this issue?

framps, 2018-04-14:
Closing because there was no feedback any more since 27 days.

closed 2018-04-14 by framps