framps/raspiBackup #43: After restore raspberry pi to a new micro card, It can not connect to internet.

caianhua1986, 2018-03-20:
I restored the 'tgz' format file to my new micro sdcard, when finished . put the new micro sdcard into raspberry pi3 . After it power on , the screem show : "failed to start dhcpcd on all interfaces". and then I execute command "sudo systemctl disable dhcpcd.service" ,reboot .  the raspberry can't connect internet.
I think if there are some stuff caused inconsistent backup.

caianhua1986, 2018-03-20:
another shell script is very good ,solved my ploblem. see: https://github.com/billw2/rpi-clone

framps, 2018-03-20:
I detected exactly this issue when I executed regression tests with Jessie. tar backup worked fine until Wheezy. I tried to find the root cause for the issue [(See this thread for details)](https://raspberrypi.stackexchange.com/questions/68859/cloned-jessie-image-fails-to-acquire-a-dhcp-address) and it's somehow related to tar, Jessie and dhcp. I finally gave up.

Your're the first one who also reports this issue. Others didn't face it. I think it may be also related to the dhcp server and it's configuration you're using in your local net.

I see following options for you:

1. Use rsync backup instead of tar backup with raspiBackup
2. Setup your network with networkd instead of dhcpcd
3. Try to find the root cause for the issue (you can add additional tar options in the raspiBackup config)
file
4. Stick with the tool you mentioned above if you don't need the additional features provided by raspiBackup

closed 2018-03-20 by framps