framps/raspiBackup #44: Dash (-) in -a or -o parameter string leads to "option required"

christianTF, 2018-03-26:
Example commandline:
` sudo /usr/local/bin/raspiBackup.sh -F -o "service avahi-daemon stop" -a ":" -k 3 -t ddz -L current -l debug -m detailed -v /backup`
leads to the error
`??? RBK0090E: Option -o requires a parameter.`

This is caused by the dash (-) in the avahi-daemon string.
The same command with `service cron stop` works as expected.
The issue also regards to the -a parameter.

lg, Christian

framps, 2018-03-26:
This is an already known issue of the initial v 0.6.3.2 published (sha 82629da). It was fixed two days later and the current v 0.6.3.2 available for installation (sha 5659ee1) no longer faces this issue. Please download the current code available on https://www.linux-tips-and-tricks.de/raspiBackup.sh.

christianTF, 2018-03-26:
You're right - already fixed!
Thanks!

closed 2018-03-26 by christianTF

framps, 2018-03-26:
Just for the records: v 0.6.3.2 got a new option parser which checked whether mandatory option parameters were passed. Unfortunately parameter strings which include a dash were identified as new options instead of option parameters.