framps/raspiBackup #46: Testing in VM

danielandross, 2018-04-14:
Before using the script on my precious pi projects I wanted to try it on a dummy VM first. It seems that it requires a /dev/sd to be present. Is the script really only applicable on pi?

framps, 2018-04-18:
@danielandross Do you need any additional information? Otherwise please close the issue ;-)

framps, 2018-04-22:
@danielandross I'm going to close the issue in one week if you don't mind :smile:

heikoh81, 2018-05-04:
> There exist already a lot of tools to backup a normal Linux system.

Could you please provide some links?
I have several Raspis at home, but I also like to get a backup of my Debian VPS.

framps, 2018-05-04:
Not sure I get your point. But if you are looking for Linux backup tools just search for `linux backup` using your favorite search engine ;-) 
For example I just found [url=https://www.tecmint.com/linux-system-backup-tools/]this page[/url] with `20 Outstanding Backup Utilities for Linux Systems in 2018`

heikoh81, 2018-05-04:
Well, I like the raspibackup script very much.
My VPS provider (Netcup) has a strange Backup model.
You can take snapshots via Webinterface, but only 1 export overall is free. Each additonal export costs 0,5€.

I'm wondering if there is a script like raspibackup that can create an image file of the running VPS and save it in the same filesystem. Then I could download it with WinSCP.

Restore of a running system would be impossible though, I guess?

framps, 2018-05-04:
Ok. I got your point. I don't know of any Linux tool which creates backups of a running system. 

We're off topic now ...