framps/raspiBackup #48: Unable to detect boot device

luisspuerto, 2018-04-24:
Hi, 

I'm trying to use raspiBackup but I get this error: 

```sh 
lpuerto@opl-cubox:~$ sudo raspiBackup.sh -a : -o : -m detailed
--- RBK0009I: opl-cubox: raspiBackup.sh V0.6.3.2a (1d77d56) started at Tue Apr 24 13:39:54 EEST 2018.
--- RBK0128I: Using logfile /backup/opl-cubox/opl-cubox-rsync-backup-20180424-133954/opl-cubox-backup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
??? RBK0142E: Unable to detect boot device.
--- RBK0026I: Saving logfile in /home/lpuerto/opl-cubox-raspiBackup.log.
--- RBK0105I: Deleting new backup directory /backup/opl-cubox/opl-cubox-rsync-backup-20180424-133954.
??? RBK0005E: Backup failed. Check previous error messages for details.
```
Instead of using a USB as backup device I'm using a folder in an already mounted hard drive, so: 

```
# path to store the backupfile
DEFAULT_BACKUPPATH="/srv/dev-disk-by-label-OpL1TB/Backups"
```
Do you know what could be the problem here? 

I'm not using it on raspbian, but in armbian.

framps, 2018-04-24:
Hello Luis,

as you know raspiBackup does not support armbian officially because I don't run armbian and nobody volonteered to test raspiBackup on armbian. But in general it should work. If you add additional options
`-l debug -L current`
a debug log will be created by raspiBackup. Please either attach it in this issue (make sure there is no sensitive info in the log) or send it to my [eMail address](https://www.linux-tips-and-tricks.de/en/contact).

luisspuerto, 2018-04-24:
```
20180424-202644: DBG  -- LOG_OUTPUT: 3
20180424-202644: DBG  -- Using logfile /home/lpuerto/raspiBackup.log
20180424-202644: DBG  -- raspiBackup.sh 0.6.3.2a, 2018-04-15/12:42:27 - 1d77d56
20180424-202644: DBG  -- BACKUPTARGET_DIR: /srv/dev-disk-by-label-OpL1TB/Backups/opl-cubox/opl-cubox-rsync-backup-20180424-202644
20180424-202644: DBG  -- BACKUPTARGET_FILE: /srv/dev-disk-by-label-OpL1TB/Backups/opl-cubox/opl-cubox-rsync-backup-20180424-202644/opl-cubox-rsync-backup-20180424-202644
20180424-202644: DBG     >> logOptions
20180424-202644: DBG     -- PRETTY_NAME="Debian GNU/Linux 9 (stretch)"
NAME="Debian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
20180424-202644: DBG     -- 9.4
20180424-202644: DBG     -- Linux opl-cubox 4.14.14-cubox #3 SMP Thu Jan 25 15:58:45 CET 2018 armv7l GNU/Linux
20180424-202644: DBG     -- Options:  -a : -o : -m detailed -l debug -L current
20180424-202644: DBG  << logOptions
20180424-202644: DBG  -- APPEND_LOG=0
20180424-202644: DBG  -- BACKUPPATH=/srv/dev-disk-by-label-OpL1TB/Backups
20180424-202644: DBG  -- BACKUPTYPE=rsync
20180424-202644: DBG  -- CHECK_FOR_BAD_BLOCKS=0
20180424-202644: DBG  -- CONFIG_FILE=
20180424-202644: DBG  -- DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20180424-202644: DBG  -- DD_BLOCKSIZE=1MB
20180424-202644: DBG  -- DD_PARMS=
20180424-202644: DBG  -- DEFAULT_DEPLOYMENT_HOSTS=
20180424-202644: DBG  -- DEFAULT_YES_NO_RESTORE_DEVICE=loop
20180424-202644: DBG  -- EMAIL=
20180424-202645: DBG  -- EMAIL_PARMS=
20180424-202645: DBG  -- EXCLUDE_LIST=
20180424-202645: DBG  -- EXTENSIONS=
20180424-202645: DBG  -- FAKE=0
20180424-202645: DBG  -- HANDLE_DEPRECATED=1
20180424-202645: DBG  -- KEEPBACKUPS=3
20180424-202645: DBG  -- LANGUAGE=EN
20180424-202645: DBG  -- LINK_BOOTPARTITIONFILES=0
20180424-202645: DBG  -- LOG_LEVEL=1
20180424-202645: DBG  -- LOG_OUTPUT=3
20180424-202645: DBG  -- MAIL_ON_ERROR_ONLY=0
20180424-202645: DBG  -- MAIL_PROGRAM=mail
20180424-202645: DBG  -- MSG_LEVEL=1
20180424-202645: DBG  -- NOTIFY_UPDATE=1
20180424-202645: DBG  -- PARTITIONBASED_BACKUP=0
20180424-202645: DBG  -- PARTITIONS_TO_BACKUP=*
20180424-202645: DBG  -- RESTORE_DEVICE=
20180424-202645: DBG  -- ROOT_PARTITION=
20180424-202645: DBG  -- RSYNC_BACKUP_ADDITIONAL_OPTIONS=
20180424-202645: DBG  -- RSYNC_BACKUP_OPTIONS=-aHAx
20180424-202645: DBG  -- SKIPLOCALCHECK=0
20180424-202645: DBG  -- STARTSERVICES=:
20180424-202645: DBG  -- STOPSERVICES=:
20180424-202645: DBG  -- TAR_BACKUP_ADDITIONAL_OPTIONS=
20180424-202645: DBG  -- TAR_BACKUP_OPTIONS=-cpi
20180424-202645: DBG  -- TAR_BOOT_PARTITION_ENABLED=0
20180424-202645: DBG  -- TAR_RESTORE_ADDITIONAL_OPTIONS=
20180424-202645: DBG  -- VERBOSE=0
20180424-202645: DBG  -- ZIP_BACKUP=0
20180424-202645: DBG  -- RESIZE_ROOTFS=1
20180424-202645: DBG  -- TIMESTAMPS=0
20180424-202645: DBG  -- RSYNC_IGNORE_ERRORS=
20180424-202645: DBG  -- TAR_IGNORE_ERRORS=
20180424-202645: DBG  -- USE_HARDLINKS=1
20180424-202645: MSG  --- RBK0128I: Using logfile /home/lpuerto/raspiBackup.log.
20180424-202645: DBG  -- /etc/config
#
# Default configuration file for raspiBackup.sh
#
# (C) 2013,2018 framp at linux-tips-and-tricks dot de
[... 225 lines cut ...]
/dev/sda
/dev/sda1
20180424-202646: DBG           -- mountpoint /boot: /boot is not a mountpoint mountpoint /: 179:1
20180424-202646: DBG           -- part: 
20180424-202646: DBG           -- deviceNumber: /boot is not a mountpoint
20180424-202646: MSG           ??? RBK0142E: Unable to detect boot device.
20180424-202646: DBG              >> exitError 102
20180424-202646: DBG           << exitError 102
20180424-202646: DBG              >> cleanup
20180424-202646: DBG                 >> cleanupBackup
20180424-202646: DBG                 -- Got trap EXIT
20180424-202646: DBG                 -- rc: 102
Invocation parms: ' -a : -o : -m detailed -l debug -L current'
20180424-202646: DBG                 -- emailTitle: opl-cubox: Backup failed !!!.
20180424-202646: DBG                    >> cleanupBackupDirectory
20180424-202646: DBG                    -- BackupDir created: 1
20180424-202646: MSG                    --- RBK0026I: Saving logfile in /home/lpuerto/raspiBackup.log.
20180424-202646: MSG                    --- RBK0105I: Deleting new backup directory /srv/dev-disk-by-label-OpL1TB/Backups/opl-cubox/opl-cubox-rsync-backup-20180424-202644.
20180424-202646: MSG                    ??? RBK0005E: Backup failed. Check previous error messages for details.
20180424-202646: DBG                 << cleanupBackup
20180424-202646: DBG                    >> cleanupTempFiles
20180424-202646: DBG                    -- Removing mailfile /tmp/raspiBackup.maillog
20180424-202646: DBG                 << cleanupTempFiles
20180424-202646: DBG                 -- Terminate now with rc 102
```

framps, 2018-04-24:
Thank you very much for the debug log. Unfortunately I just noticed there is are important debug statements missing in raspiBackup :unamused: 

Please show the output of following commands:
`sudo findmnt /boot -o source -n`
`sudo ls -la /boot`

luisspuerto, 2018-04-25:
perhaps this helps: 

```
lpuerto@opl-cubox:~$ sudo findmnt / -o source -n
/dev/mmcblk0p1
```

framps, 2018-04-25:
Unfortunately raspiBackup requires two partitions right now :-(. armbian uses only one partition. I created [an enhancement request ](https://github.com/framps/raspiBackup/issues/49) and next release will support armbian.

luisspuerto, 2018-04-25:
Thanks a lor @framps!

luisspuerto, 2018-04-25:
I think I shall close the issue, shan't I?

framps, 2018-04-25:
Yep :smile:

closed 2018-04-26 by luisspuerto