framps/raspiBackup #49: Support operating system which use one partition only with /boot and / in normal backup mode

framps, 2018-04-25:
There exist operating systems like armbian or ubuntu which don't have two partitions (/boot and /) and just one with /boot and /.

Right now normal backup mode of raspiBackup requires two partitions.

framps, 2018-07-15:
@luisspuerto The boot device discovery was rewritten. Would be great if you can test the fix (see https://github.com/framps/raspiBackup/issues/53 for details)

luisspuerto, 2018-07-15:
@framps great! I'm out of town on vacation and I'll not return till next month. I'll test it them 😄

framps, 2018-08-11:
Closing this as a duplicate of #53 

Please use the [development branch](https://github.com/framps/raspiBackup/tree/development) for verification

closed 2018-08-11 by framps