framps/raspiBackup #54: raspiBackup log file creates rsync and tar errors because it's also saved

framps, 2018-06-04:
Right now the raspiBackup log can be stored at

1. /var/log/syslog
2. /var/log/raspiBackup/<hostname>.log
3. <backupPath>
4. ./raspiBackup.log

logs 1,2 and 4 are included in the backup process and may create tar or rsync error messages (file changed during backup). Even for 3 this ist true because the log file ist stored in ~ and moved to <backuppath> when terminating.

Workaround: Exclude the log file (use ~/raspiBackup.log with 3) with option -u

framps, 2018-07-01:
There is a beta available which includes a fix for this issue. Feel free to grab the [beta](https://github.com/framps/raspiBackup/tree/beta) and verify the fix.

McNose, 2018-07-05:
Hi,

first of all thank you for that great program.
Unfortunately I get the RC1 error. I tried the bate version as well with no luck. The backup crashes close to the end.
The error message is: RBK0021E: Backupprogramm des Typs dd beendete sich mit RC1.

Do you have any idea?
Maybe the workaround will help, but I don´t unterstand how to use it.

Thanks a lot
McNose

framps, 2018-07-05:
Sounds do me there is already an issue with the SD card :cry: . Please add option `-l debug` and start raspiBackup again. Then attach the debug log to this issue and I'll check the log..

McNose, 2018-07-05:
Thanks you for helping me.

I copied the text from the log-file into a txt. But its content is a 100% the same.

Maybe it might be important that I am using a NAS to backup the image?

Thank you.
[RaspiBackup.log](https://github.com/framps/raspiBackup/files/2168065/RaspiBackup.log)

McNose, 2018-07-06:
Thank you for your advince concerning the credentials. I am not sure what to do, but I will try.

I also tried the `touch "/backup/raspberrypi/dummyFile"` as root. It worked. I could generate the file on my NAS.
After that I tried to backup again using the root user on the pi. Still the same error.

I will will try a backup on an USB-Stick tomorrow. But if you have any further ideas, I would be thankful.

McNose, 2018-07-06:
Maybe it help to know, that there is a backup file on my NAS. I can see it in my recycle bin.

framps, 2018-07-06:
I guess there is something weird in following mount options.
`//192.168.178.88/Backup on /backup type cifs (rw,relatime,vers=1.0,cache=strict,username=admin,domain=,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.178.88,soft,unix,posixpaths,serverino,mapposix,acl,rsize=1048576,wsize=1048576,echo_interval=60,actimeo=1,_netdev)`
It's a very good idea :+1: to test to create the backup on an USB disk or stick. That way you test whether my guess is right.

McNose, 2018-07-06:
Yes. Your guess is right!
It worked on an usb stick.

But what might be the problem with the mount options?
I can not understand where all the options come from you see.

I mount the device using: `sudo mount -t cifs -o vers=1.0,_netdev,username=name,password=12345 //192.168.178.88/Backup /backup/`

I dont know what `-t cifs -o` does, but it seems to be the normal way.
I added `vers=1.0,_netdev`, what I found somewhere in the internet, because otherwise I am always told that the host is down.

Using my command the mount seems to work. I can see the mounted files on the raspberry. But it looks like the version=1.0,_netdev might be the problem. Right?

framps, 2018-07-07:
I frankly mount all my stuff via nfs. But you're not the first one who reported issues with mounting a samba drive on a Raspi and therefore have an issue with raspIBackup. But your issue is a samba config issue but no raspiBackup issue :wink: Samba is hard to configure. It also depends on the NAS system you use and it's configuration. :cry: 

I suggest to ask the community for help. Either use the [German forum](https://forum-raspberrypi.de/forum/) or the [english forum](https://www.raspberrypi.org/forums/). Would be great if you append the solution in this issue - assuming you find the solution

McNose, 2018-07-07:
My NAS can use nfs too.
Maybe I will give that a try.

If I find a solution I will add it here.
Thanks you so far.

framps, 2018-07-07:
nfs is faster than samba and also supports hardlinks which is very useful for rsync backups (that's what I use). There is a page about [synonolgy and nfs](https://www.linux-tips-and-tricks.de/en/synology-usage) which may help you on your nfs setup :wink:

McNose, 2018-07-07:
Thank you for your help. Unfortunately the link does not help. I got the error:
`mount.nfs: mount system call failed`

I tried your link and these two [(1](https://kodi.wiki/index.php?title=NFS#Synology), [2](https://www.elektronik-kompendium.de/sites/raspberry-pi/2102211.htm)). Maybe you can still help here.

framps, 2018-07-07:
I help with all issues with raspiBackup and I also help as long as it takes me a minute for other issues. Your issue is not a raspIBackup issue and I sent you links I'm aware of which may help.

Please touch base with the community now. If you decide to visit either forum mentioned above please add a link to your post her and I'll subscribe your thread  and help if any raspiBackup specific information is asked for.

Please note: Everything I do is for free and I would be busy all day long if I help everybody who faces environment or configuration issues just because he tries to use raspiBackup :wink:

framps, 2018-09-25:
A user discovered with 0.6.4 the logfile is still written in /root.
```
root@opizero:/usr/local/etc# lsof / | awk 'NR==1 || $4~/[0-9][uw]/'
COMMAND     PID       USER   FD   TYPE DEVICE SIZE/OFF   NODE NAME
dhclient   1036       root    4w   REG  179,1     6636  37408 /var/lib/dhcp/dhclient.eth0.leases
tee        1248       root    3w   REG  179,1    20086   1113 /root/opizero-raspiBackup.log
tee        1250       root    3w   REG  179,1    20086   1113 /root/opizero-raspiBackup.log
```

framps, 2018-10-13:
Fix was merged in 0.6.4.1 beta

closed 2019-01-12 by framps