framps/raspiBackup #57: Restore to smaller card does not work anymore

giase82, 2018-06-29:
... is this still supposed to work? The FAQ states:
"If the target SD card is smaller or larger than the original SD card the second root partition will be adjusted accordingly"

I'd like to restore a 7GB tar-backup which has been created from a 64GB SD card to a 16GB SD card. 

`
sudo raspiBackup.sh -d /dev/sdb pi-tar-backup-20180628-201457/
--- RBK0009I: cfers-ubuntu-vm-01: raspiBackup.sh V0.6.3.2a (1015c57) started at Fr 29. Jun 08:39:10 CEST 2018.
--- RBK0128I: Using logfile /home/.../Downloads/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
??? RBK0160E: Target /dev/sdb with 14.45 GiB is smaller than backup source with 59.47 GiB.
??? RBK0077E: Restore failed with RC 102. Check previous error messages.
`

Or am I doing something wrong?

Thanks for your help!

giase82, 2018-06-29:
Hi!

Thanks for your swift response! 

Are partitioned backups in general not working or is there special settings?

Just to try this out, I commented out the line which throws the error and exits, and it worked like a charm! Is this expected?
I think I have not changed settings and those backups worked for being restored... 

Cheers!

framps, 2018-06-29:
Not sure I get your point. Do you use partitioned backup? If not please use the original raspiBackup version and pass the debug log to me for further investigations. Either in this issue or send it to me (See my eMail on the contact page)

giase82, 2018-07-02:
Hi, sorry for the confusion. I've been on the road while writing.

Yes, I am using a partitioned backup, and I believe it was possible to restore such a backup, even if the size is different. Is that true or am I simply wrong.

In addition to that, just for fun, I tried out commenting this line here:
`
	if (( targetSDSize < sourceSDSize )); then
		writeToConsole $MSG_LEVEL_MINIMAL $MSG_TARGETSD_SIZE_TOO_SMALL "$RESTORE_DEVICE" "$(bytesToHuman $targetSDSize)" "$(bytesToHuman $sourceSDSize)"
		**#exitError $RC_MISC_ERROR**
`

I understand that this is not absolutely right, but it seems the restore still worked. So I wonder if there is a possibility to restore to a smaller card for partitioned backups using tar?

framps, 2018-07-02:
I added the partitionoriented backup mode because people asked for this feature. It took me a lot of time to implement and test it and I finally decided not to spend more time on this to get the shrink function working because that's much more complex than for the normal backup. 

It's only about 10% of the raspiBackup users use partition oriented mode. That's a very low number of users. I'll fix issues with the partition oriented mode but will not add any new feature for this mode. It's not worth to add any new feature for partition oriented backup. Keep in mind I'll work on raspiBackup in my spare time :wink:

giase82, 2018-07-02:
Sure, no problem! I just wanted to understand a bit better. I am currently working on changing my setups away from NOOBS (which is just a left-over from times when I started). I hope I manage to "convert" from NOOBS to normal. Seems a bit tricky but doable.

Thanks a lot for your work on raspiBackup!

Cheers,
//Christof

framps, 2018-07-02:
It's a very good decision to go away from NOOBs :+1:  ([See my thoughts on this](https://www.linux-tips-and-tricks.de/en/noobs-support)). 

Please close this issue if you're fine now.

closed 2018-07-02 by giase82