framps/raspiBackup #59: Backup aborts with "unknown command" for tar

giase82, 2018-07-03:
Hi,

My backup aborts as there seems to be the wrong tar-command used:

### This is the error:
--- RBK0009I: shpihuber: raspiBackup.sh V0.6.3.2a (1015c57) started at Di 3. Jul 08:57:15 CEST 2018.
--- RBK0128I: Using logfile /home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714/shpihuber-backup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0151I: Using backuppath /home/smarthome/mnt/remote_home/raspibackup.
!!! RBK0157W: No services to stop.
--- RBK0081I: Creating backup of type tar in /home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714.
--- RBK0036I: Saving partition layout.
--- RBK0044I: Creating backup of boot partition in /home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714/shpihuber-backup.img.
--- RBK0045I: Creating backup of partition layout in /home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714/shpihuber-backup.sfdisk.
--- RBK0046I: Creating backup of master boot record in /home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714/shpihuber-backup.mbr.
--- RBK0158I: Creating native tar backup "/home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714/shpihuber-tar-backup-20180703-085714.tar".
--- RBK0085I: Backup of type tar started. Please be patient.
**??? RBK0024E: Backup tool tar received error 132. Errormessages:
/usr/local/bin/raspiBackup.sh: Zeile 1119:  5887 Ungültiger Maschinenbefehl   tar -cpi -f "/home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714/shpihuber-tar-backup-20180703-085714.tar" --one-file-system --warning=no-xdev --numeric-owner --exclude="/home/smarthome/mnt/remote_home/raspibackup/*" --exclude="/home//shpihuber-raspiBackup.log" --exclude='.gvfs' --exclude=proc/* --exclude=lost+found/* --exclude=sys/* --exclude=dev/* --exclude=tmp/* --exclude=boot/* --exclude=run/* /**
??? RBK0021E: Backupprogram for type tar failed with RC 132.
--- RBK0026I: Saving logfile in /home//shpihuber-raspiBackup.log.
--- RBK0043I: Removing incomplete backup in /home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714. This will take some time. Please be patient.
--- RBK0105I: Deleting new backup directory /home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714.
??? RBK0005E: Backup failed. Check previous error messages for details.

[shpihuber-raspiBackup.log](https://github.com/framps/raspiBackup/files/2157989/shpihuber-raspiBackup.log)

framps, 2018-07-03:
That's a weird error message. It's syntactical OK. I executed it on my system by faking some files and tar start without any issue.
>/home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714/shpihuber-tar-backup-20180703-085714.tar"

I notice you mounted your backup in your home directory which is very uncommon. But I don't think that's the root cause.

Because I cannot reproduce he issue and the debug log also doesn't help. So I suggest to nail down the issue on your environment:

Execute the command without all the options
`sudo tar -cpi -f "/home/smarthome/mnt/remote_home/raspibackup/shpihuber/shpihuber-tar-backup-20180703-085714/shpihuber-tar-backup-20180703-085714.tar"  /`

If it works remove one option step by step until it fails. Please keep the leading /.

Let me know when you get tar starting without an issue.

giase82, 2018-07-05:
It is weird... I put the same config on a different Raspi and this just works.
I assume it must be a broken SD card or something like this.

Closing here therefore.

Thanks anyways!

closed 2018-07-05 by giase82