framps/raspiBackup #60: DD backup on SD card does not boot

dersch81, 2018-07-12:
Hi,

OS: jessie

I have a problem with booting my backup. I used dd for backing up and tried with Win32DiskImager and with a native debian "dd if=file.img of=/dev/sdb" but the pi does not boot and just shows a black screen. If i check the img file there are 4 partitions inside and all files are there. On the sdcard after pushing the img on it are 3 partitions: boot, root0 and settings with all files of the backuped system. 

How can i verify that the pi will boot from the sdcard?

framps, 2018-07-12:
> On the sdcard after pushing the img on it are 3 partitions: boot, root0 and settings with all files of the backuped system.
1. Please plug in the backup image into your system you want to backup and provide the output of `sudo fdisk -l`. Then I can check which partitions are defined on source and backup.
2. What do you mean with 
> settings with all files of the backuped system.

dersch81, 2018-07-13:
I could solve the issue with the DD option bs=4M. Right after that the sdcard has been created with the right partition table.

closed 2018-07-13 by dersch81

framps, 2018-07-13:
Whow. Thank you very much for the update. :+1: I wouldn't have thought this is the root cause for your issue.