framps/raspiBackup #64: Möglichkeit das Übergabeparameter in Extension übergeben werden

Hofei90, 2018-08-08:
Hallo Framp,
ich hoffe mal dass ich es im Benutzerhandbuch nicht überlesen habe, aber gibt es eine Möglichkeit dass raspiBackup der Extension beim Aufruf Parameter übergibt.

Wenn nein seh diese Frage bitte als Feature Request an.
Im konkreten hätte ich gerne den Pfad und Dateiname als Parameter zur Verfügung wenn die Extension aufgerufen wird. Ziel des ganzen ist, dass ich pishrink vor dem Packen dazwischen schalten möchte und dann anschießend das ganze erst mit zip packen. Die komfortabelste Lösung darin sehe ich, wenn mir raspiBackup eben den Dateiname und Pfad gleich mitteilt und ich dies nicht manuell ermitteln muss.

framps, 2018-08-08:
Moin @Hofei90,

Du willst also ein dd Backup erstellen, dann pishrinken und dann zippen?

Dazu ist das Wrapperscript für dich bestens geeignet. Du musst den Aufruf von pishrink enablen und anschließend zip auf das Ergebnis starten. 

Das sollten nur ein paar Zeilen sein die Du zufügen musst.

Alternativ kannst du auch raspiBackup benutzen um nur die allokierte Partition zu sichern. Siehe faq 16. Allerdings ohne pishrink.

Cu framp

Hofei90, 2018-08-08:
FAQ 16 erscheint mir zu unpraktikabel, da müsst ich ja zuvor immer Hand anlegen um die Partition zu verkleinern...
Das mit dem pshrink enablen denke ich ist noch ganz leicht, das ist wohl nur Zeile 180 einzukommentieren, was mir aber noch nicht klar ist, wie greife ich direkt auf das Ergebnis über das Skript zu um zu zippen?
https://github.com/framps/raspiBackup/blob/master/helper/raspiBackupWrapper.sh#L180

framps, 2018-08-09:
Weiter oben steht 
```
 now following variables are available for further backup processing
# BACKUP_TARGETDIR refers to the backupdirectory just created
# BACKUP_TARGETFILE refers to the dd backup file just created
```
D.h. Du kannst die Variable $BACKUP_TARGETFILE benutzen um auf das dd File zuzugreifen. 

D.h. es sollte reichen die letzten beiden Zeilen
```
#pishrink
#raspiBackupRestore2Image
```
in
```
pishrink
gzip "${BACKUP_TARGETFILE}
```
zu ändern.

Ist alles aber ungetestet :wink:

Hofei90, 2018-08-12:
Ab jetzt ist es getestet und funktioniert :)
Danke für die Hilfe!

closed 2018-08-12 by Hofei90