framps/raspiBackup #65: Änderung Verzeichnisstruktur

Hofei90, 2018-08-12:
Im Bezug auf das Benutzerhandbuch Kapitel Backupverzeichnisstruktur (Normaler Backup)
https://www.linux-tips-and-tricks.de/de/raspibackup#verzeichnis
(nebenbei bemerkt: Rechtschreibfehler 2. Wort -> Backplauf --> Backuplauf

Aktuell habe ich für jeden meiner Pis einen eigenen Ordner auf der NAS erstellt der aus Mount Punkt dient.

mein mountpunkt am pi ist: /media/fritz_nas
Eingehängt wird: //192.168.178.1/fritzbox/Sicherung/RaspberryPi/Lernumgebung
darin befindet sich der Ordner backup in dem von raspiBackup gesichert werden soll, somit die Angabe bei raspiBackup:
# Pfad wo das Backupfile gespeichert wird
DEFAULT_BACKUPPATH="/media/fritz_nas/backup"

allerdings entsteht jetzt auf der nas so eine Ordnerstruktur (wie ja auch in denem Benutzerhandbuch angegeben ;) ):
B:\Sicherung\RaspberryPi\Lernumgebung\backup\Lernumgebung\Lernumgebung-dd-backup-20180812-163257

nach dem Ordner backup der Ordner Lernumgebung...den will ich weg habe, nur find ich nicht die Einstellmöglichkeit desssen, hab ich überhaupt die Möglichkeit dazu, dass raspiBackup den Ordner <hostname> nicht mit anlegt?

framps, 2018-08-12:
Vielen Dank fuer den Hinweis auf den Typo. Ist schon korrigiert. Ist aber garantiert nicht der Letzte :smile: 

raspiBackup erstellt immer ein Basisverzeichnis welches dem Hostnamen entspricht. D.h. bei Dir ist es `Lernumgebung`. Das ist das Stammverzeichnis. Darin werden dann die Backups in Unterverzeichnissen abgelegt wobei der Name sich immer ergibt aus  der Hostname gefolgt vom Backuptyp und Datum und Zeit., bei Dir also `Lernumgebung-dd-backup-20180812-163257`.

Dieses Namensschema ist fest und kann nicht geändert werden. Wenn Du eine andere Verzeichnisstruktur haben willst musst Du sie Dir parallel mit Softlinks erstellen oder die Dateien nach einem Backup umkopieren. Das kannst Du z.B. im Wrapperscript oder in einem Plugin tun.

Hofei90, 2018-08-12:
Wollte das ganze jetzt mit folgender Zeile nach dem gzip Teil im raspiWrapper Skript lösen:
`mv "${BACKUP_TARGETDIR}" "${BACKUP_PATH}/"
`
Hierbei produziere ich aber folgende Fehlermeldung:
mv: der Aufruf von stat für '' ist nicht möglich: Datei oder Verzeichnis nicht gefunden

warum wird dadurch ein leeres Statement ausgegeben? Das ist doch das selbe Schema wie beim gzip Teil
gzip "${BACKUP_TARGETFILE}" , wohingegen das mit gzip funktioniert.

Hofei90, 2018-08-13:
Hier das Skript:
[raspiBackupWrapper.zip](https://github.com/framps/raspiBackup/files/2282063/raspiBackupWrapper.zip)
Auszug dessen:
`BACKUP_MOUNT_POINT="/media/fritz_nas"							 # ===> adapt to your environment
BACKUP_PATH="$BACKUP_MOUNT_POINT/backup"              # ===> adapt to your environment`
Wenn ich diese Konstellation am Terminal nachsimuliere würde ich eigentlich zum passenden Ergebnis kommen.

framps, 2018-08-13:
Irgendwas stimmt da nicht :kopfkratz:  Schreibe mal 
`set -x`
in Zeile 165 und sieh Dir an wie der mv Befehl zur Laufzeit genau aussieht.

Hofei90, 2018-08-14:
Ausgabe von set -x

> Backup succeeded :-)
+ pishrink
+ which pishrink.sh
+ readVars
+ [[ -f /tmp/raspiBackup.vars ]]
+ source /tmp/raspiBackup.vars
++ BACKUP_TARGETDIR=/media/fritz_nas/backup/Lernumgebung/Lernumgebung-dd-backup-20180814-114752
++ BACKUP_TARGETFILE=/media/fritz_nas/backup/Lernumgebung/Lernumgebung-dd-backup-20180814-114752/Lernumgebung-dd-backup-20180814-114752.img
+ pishrink.sh /media/fritz_nas/backup/Lernumgebung/Lernumgebung-dd-backup-20180814-114752/Lernumgebung-dd-backup-20180814-114752.img

Das verwirrende, mit dem Einbau von set -x funktioniert das ganze jetzt.

framps, 2018-08-14:
Jetzt bin ich verwirrt. Bei Dir fehlt die Debugausgabe von
```
gzip "${BACKUP_TARGETFILE}"
mv "${BACKUP_TARGETDIR}" "${BACKUP_PATH}/" 
```
:kopfkratz:

Benutzt Du das nicht mehr? In Deinem zip File steht es aber drin. Wie kann dann aber alles funktionieren? Was ist alles? Mit oder ohne gzip? 

Am `set -x` kann es jedenfalls nicht liegen.

Hofei90, 2018-08-16:
So sry, gestern war Pi freie Zeit ;)
Doch benutz ich schon noch, Skript wurde so ausgeführt wie in der zip.
Hätte die Ausgaben jetzt mal in eine Datei umgeleitet, diese enthält aber wieder nicht die Ausgaben von set -x ?!
```
--- /media/fritz_nas already mounted
Backup succeeded :-)
Creating new /etc/rc.local
rootfs: 41308/940576 Dateien (0.2% nicht zusammenhängend), 358318/3890944 Blöcke
Die Größe des Dateisystems auf /dev/loop0 wird auf 420474 (4k) Blöcke geändert.
Start von Durchgang 2 (max = 19323)
Blöcke werden verschoben     XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Start von Durchgang 3 (max = 119)
Die Inode-Tabelle wird gelesenXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Start von Durchgang 4 (max = 4010)
Die Inode-Referenzen werden aktualisiertXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
Das Dateisystem auf /dev/loop0 is nun 420474 (4k) Blöcke lang.

Shrunk /media/fritz_nas/backup/Lernumgebung/Lernumgebung-dd-backup-20180816-191045/Lernumgebung-dd-backup-20180816-191045.img from 15G to 1,7G
pishrink succeeded :-)
```
Da aber das verschieben wieder erfolgreich war, sehen wir das Thema als erledigt an?! Ich bin zumindest aktuell zufrieden ;)

framps, 2018-08-16:
Sorry - ich verstehe Dich nicht :cry:  Aber wenn es jetzt funktioniert wie gewuenscht dann kannst Du den Issue ja closen :smile:

Hofei90, 2018-08-16:
> Benutzt Du das nicht mehr? In Deinem zip File steht es aber drin. Wie kann dann aber alles funktionieren? Was ist alles? Mit oder ohne gzip?

Ja das Image wurde auch gzipt, allerdings erschien es nicht in der Ausgabe welche ich gepostet habe. Warum auch immer...

Jedenfalls es funktioniert jetzt, warum es beim ersten mal nicht ging kann ich leider nicht mehr rekonstruieren.
So hoff das was jetzt besser formuliert. 
Danke für die Hilfe! raspiBackup nimmt auf 3 weiteren Pis seinen Dienst auf ;)

closed 2018-08-16 by Hofei90