framps/raspiBackup #69: Output to /dev/tty

christianTF, 2018-09-11:
Hi again,

it took me some time, tests, and finally code review to figure out, that your output is sent to `/dev/tty`.
Also it is tricky to capture that output from tty.

Is there a special reason to not use stderr or stdout?

cu, Christian

framps, 2018-09-12:
Well, that's something I coded in the early development stage of raspiBackup. As far as I remember it was somehow related to the fact that the output is written in parallel into the log. 

What's your exact use case? I guess you want to pipe the output somewhere in your code when invoking raspiBackup. @christianTF Please confirm. 

I will check whether I can change the code to use stdout and stderr for all messages.

christianTF, 2018-09-12:
As the LoxBerry uses native logging facilities with live web logviewer, I tried to pipe the output directly to the LogBerry logging instead of copying it _after_ the backup. 

Finally I found the tool `script —command raspiBackup.sh ...` that captures tty and send it to my file.

This is a working solution where I don’t require any adaption/feature. It too would be compatible with stdout in the case you are thinking of a modification.

I was just asking because I’ not familiar with tty output.

Cheers, Christian

framps, 2018-09-12:
Glad you found a solution for your issue. I suggest to create an issue first next time you face some strange behavior of raspiBackup before spending time to find a workaround :wink:.

Usage of tty does not comply to standard io handling of programs. I frankly didn't care about this in early development stages. But you put a valid issue on the table. I found out it's not that difficult to redirect I&W messages into stdout and E messages into stderr.
```
sudo ./raspiBackup.sh -l debug -L current 1>stdout.log 2>stderr.log
```
works already in a prototype. Is this what you are looking for and will help you to get rid of your workaround?

I keep you posted in this issue when there is a fix available. I will roll in this fix in next release and then it's up to you whether you want to stick with your workaround or use stdout and stderr.

christianTF, 2018-09-19:
This looks good and familiar :-)

I'll stick on this issue, and will test it as soon as you have a release with the change.

framps, 2018-09-19:
Glad to hear that's what you need :smile: 

This feature is already available in the development branch which will become the next release. Feel free to grab the code and test right now. Please note: The development branch includes all issues marked as `fixed in next release` and thus includes your two other issues you created :smile:

closed 2019-01-12 by framps