framps/raspiBackup #71: "Snapshot restore" feature für rsync/tar/tgz

christianTF, 2018-09-19:
I am investigating in creating a kind of "snapshots restore" with raspiBackup.

What I mean is a way to restore to a (normal made) rsync or tar backup _on the running device_.

The feature would be:
On a running Pi, restore all files to the state of a selected rsync/tar backup.
This kind of restore does not fix or prepare anything on the /boot or / partition, and is using the current and mounted / partition to restore.
In short, it is an rsync copy from backup to /, replacing all changed files to the backup state, and deleting all newer files.

This snapshot restore should exclude the claim to be consistent (open files,...) and to be a restore option for a dead Pi, but can restore as good as possible to an earlier state on a running system.

My "snapshot" term is borrowed from virtual environments, where a can restore a (running) VM to an earlier state with little effort and time.

The "snapshots" would be normal rsync or tar backups, and the "snapshot restore" would be the restore of these backups but without all the "partition magic" of raspiBackup.

lg, Christian

framps, 2018-09-19:
raspiBackup supports a kind of snapshot with option -M. But you need to restore the backup to a new SD card.
You are looking for a way to restore the backup without creating a new SD card.

There is one major difference between a VM snapshot and a raspiBackup:  A VM snapshot saves and restores the main memory together with the disk contents in order to create a consistent system during restore. raspiBackup is not able to save and restore the main memory. A backup restored by raspiBackup will only restore the file system and therefore create an inconsistent system.

framps, 2018-09-19:
The aim and purpose of raspiBackup is to create a consistent backup and a consistent restored image.

Why don't you create a oneliner which just uses rsync to restore the rsync backup created by raspiBackup onto the running system?

christianTF, 2018-09-21:
I will do so. 
lg, Christian

closed 2018-09-21 by christianTF