framps/raspiBackup #74: Issues with RootFS

mars-online, 2018-09-27:
I guess raspiBackup does not recognize that I use a rootoverlay. Might that be possible? I use Raspian and provided some additional information. Might this be a bug or a issue on my side? 

$ sudo /usr/local/bin/raspiBackup.sh -k 4 -t ddz -a "service apache2 start" -o "service apache2 stop" /backup
--- RBK0009I: 3hawk: raspiBackup.sh V0.6.4 (fc7698d) started at Do 27. Sep 17:54:20 UTC 2018.
??? RBK0142E: Unable to detect boot device.
--- RBK0195I: Logfile saved in /home/pi/3hawk-raspiBackup.log.
--- RBK0105I: Deleting new backup directory /backup/3hawk/3hawk-ddz-backup-20180927-175415.
??? RBK0005E: Backup failed. Check previous error messages for details.

$ df -h
df: `/boot': Input/output error
Filesystem               Size  Used Avail Use% Mounted on
/dev/root                 29G  5,0G   23G  19% /
devtmpfs                 238M     0  238M   0% /dev
tmpfs                     49M  232K   49M   1% /run
tmpfs                    5,0M     0  5,0M   0% /run/lock
tmpfs                     97M     0   97M   0% /run/shm

[..]
20180927-175713: DBG     -- mount
/dev/root on / type ext4 (rw,noatime,data=ordered)
devtmpfs on /dev type devtmpfs (rw,relatime,size=242776k,nr_inodes=60694,mode=755)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=49412k,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /run/shm type tmpfs (rw,nosuid,nodev,noexec,relatime,size=98800k)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
/dev/mmcblk0p5 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
//192.168.0.6/backupoc/ on /backup type cifs (rw,relatime,vers=1.0,cache=strict,username=pi,domain=FATHAWK,uid=1000,forceuid,gid=1000,forcegid,addr=192.168.0.6,file_mode=0644,dir_mode=0755,nounix,serverino,mapposix,rsize=61440,wsize=65536,actimeo=1)
20180927-175713: DBG        >> doitBackup 1
20180927-175713: DBG        -- TRAP cleanup SIGINT SIGTERM EXIT
20180927-175713: DBG           >> reportNews
20180927-175713: DBG              >> isUpdatePossible
20180927-175713: DBG                 >> isNewVersionAvailable
20180927-175713: DBG                 -- Versionsplit: 0.6.4 - 
20180927-175713: DBG                 -- Returning: 0.0 0.0 0.6.4
20180927-175713: DBG                 << isNewVersionAvailable - RC: 99
20180927-175713: DBG              << isUpdatePossible
20180927-175713: DBG              >> isBetaAvailable
20180927-175713: DBG              << isBetaAvailable: 
20180927-175713: DBG           << reportNews
20180927-175713: DBG           >> getRootPartition
20180927-175713: DBG           -- cat /proc/cmdline
dma.dmachans=0x7f35 bcm2708_fb.fbwidth=592 bcm2708_fb.fbheight=448 bcm2708.boardrev=0xe bcm2708.serial=0x574759d1 smsc95xx.macaddr=B8:27:EB:47:59:D1 bcm2708_fb.fbswap=1 bcm2708.uart_clock=3000000 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=/dev/sda1 rootfstype=ext4 elevator=deadline rootwait
20180927-175713: DBG           -- RootPartition: /dev/sda1
20180927-175713: DBG           << getRootPartition: /dev/sda1
20180927-175713: DBG           >> inspect4Backup
20180927-175713: DBG           -- ls /dev/mmcblk*:
/dev/mmcblk0
/dev/mmcblk0p1
/dev/mmcblk0p2
/dev/mmcblk0p3
/dev/mmcblk0p5
/dev/mmcblk0p6
20180927-175713: DBG           -- ls /dev/sd*:
/dev/sda
/dev/sda1
20180927-175713: DBG           -- mountpoint /boot: 179:5 mountpoint /: 8:1
20180927-175715: DBG           -- part: 
20180927-175715: DBG           -- bootDeviceNumber: 179:5
20180927-175715: DBG           -- rootDeviceNumber: 8:1
20180927-175715: MSG           ??? RBK0142E: Unable to detect boot device.
20180927-175715: DBG              >> exitError 102
20180927-175715: DBG              << exitError 102
[..]

framps, 2018-09-27:
Please try raspiBackup.sh from [development branch](https://github.com/framps/raspiBackup/tree/development)

framps, 2018-10-11:
@mars-online Any updates on using the dev branch? You may even use the [beta branch](https://github.com/framps/raspiBackup/tree/beta) of next release

framps, 2018-10-17:
@mars-online Would be great if you test the beta branch. I'll close the issue in 2 weeks if I don't get any response.