framps/raspiBackup #75: DEFAULT_PARTITIONS_TO_BACKUP="1 2" produces an error

JoergBo, 2018-09-29:
I tried to use partition based backup, setting this parameters in raspiBackup.conf

DEFAULT_PARTITIONBASED_BACKUP="1"
DEFAULT_PARTITIONS_TO_BACKUP="1 2"

Its a Raspi 3B with Debian stretch with actual updates
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"

When i try to start i get an error

--- RBK0009I: boox-rpi3: raspiBackup.sh V0.6.4 (e93685e) um Sat 29 Sep 08:52:23 CEST 2018 gestartet.
--- RBK0128I: Logdatei ist /media/data/backup/[...]-rsync-backup-20180929-085223/boox-rpi3-backup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
??? RBK0093E: Angegebene Partition 1 2 der Option -T existiert nicht.

When i do the same on the commandline -T "1 2" the backup works!
In the debug.log it seems, raspibackup detects the correct partitions, but stops with the error.
Please check, thank you.

[boox-rpi3-raspiBackup.log](https://github.com/framps/raspiBackup/files/2430375/boox-rpi3-raspiBackup.log)

framps, 2018-09-29:
Configuration files used by installer were just updated with additional comments and the doc for the option -T was also updated right now on https://www.linux-tips-and-tricks.de/de/raspibackup#parameter

framps, 2018-09-29:
@JoergBo Got it now :smile: . 
Looks like I have to spend more time on this issue. I frankly thought it will work with parentheses.

framps, 2018-09-30:
@JoergBo I had to change some code and a fix ix available [here](https://github.com/framps/raspiBackup/tree/fixPartitionNumbers). Please verify and when you closed the issue I will roll in the fix into the main branch.
Thank you very much for reporting this issue :+1:

JoergBo, 2018-10-01:
@framps Tested it, DEFAULT_PARTITIONS_TO_BACKUP="1 2" works now. Thank you!

JoergBo, 2018-10-01:
Solved.

closed 2018-10-01 by JoergBo