framps/raspiBackup #77: DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY is ignored

basti12203, 2018-10-17:
Hello,
I have a 16 GB Disk and the Backup is always this size but only around 1,6 GB are used.
`root@raspberrypi:/# grep SAVE_USED /usr/local/etc/raspiBackup.conf
DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY="1"

root@raspberrypi:/usr/local/bin# raspiBackup.sh -a "sleep 1" -o "sleep 1"
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.4 () started at Wed 17 Oct 08:39:53 CEST 2018.
--- RBK0128I: Using logfile /backup/ladmin/pi_backup//raspberrypi/raspberrypi-dd-backup-20181017-083952/raspberrypi-backup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0141I: Saving space of defined partitions only.
--- RBK0151I: Using backuppath /backup/ladmin/pi_backup.
--- RBK0008I: Stopping services: 'sleep 1'.
--- RBK0081I: Creating backup of type dd in /backup/ladmin/pi_backup//raspberrypi/raspberrypi-dd-backup-20181017-083952.
--- RBK0003I: Backup size will be truncated from 15.02 GiB to 15.02 GiB.
--- RBK0085I: Backup of type dd started. Please be patient.
`

The running dd command look like

`root     21769  4.9  1.0   4532  2392 pts/3    D+   08:40   0:07 dd if=/dev/mmcblk0 bs=1048576 count=15383 of=/backup/ladmin/pi_backup//raspberrypi/raspberrypi-dd-backup-20181017-083952/raspberrypi-dd-backup-20181017-083952.img`

and my Partitiontable is this

`Disk /dev/mmcblk0: 15 GiB, 16130244608 bytes, 31504384 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x7f143cdc

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    96663    88472 43.2M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 31504383 31406080   15G 83 Linux

root@raspberrypi:/# du -sh /backup/ladmin/pi_backup/raspberrypi/raspberrypi-dd-backup-20181017-083952/
16G     /backup/ladmin/pi_backup/raspberrypi/raspberrypi-dd-backup-20181017-083952/
root@raspberrypi:/#
`
So I do not understand why the backupfile is 15GB.

Best Regards

framps, 2018-10-17:
Thank you very much for providing all the detailed information. This helped me to quickly identify your issue :smiley: 

Option `DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY` forces raspiBackup to save the defined partitions only - not the whole SD card. Your root partition is 15G and thus raspiBackup will save 15GB. If your system only needs 1,6 GB you should resize the root partition to 1,6 GB plus some spare space, for example 2.3 GB, with a partitioning tool like gparted. Than raspiBackup will save 2,3 GB only :wink: 

See also [FAQ16](https://www.linux-tips-and-tricks.de/en/faq#a16)

framps, 2018-10-17:
This will not work :cry: . You have to resize the filesystem first. Keep in mind the blocks used in the partition are not stored continuous. 

But you can use [pishrink](https://github.com/Drewsif/PiShrink) to shrink an imag. If you use the [raspiBackup wrapperscript](https://github.com/framps/raspiBackup/tree/master/helper) the saved dd image will be shrinked to it's minimum size :smiley:

framps, 2018-10-23:
@basti12203 Would you please close this issue?

closed 2018-10-24 by basti12203