framps/raspiBackup #83: Raspibackup ends in between, afterwards no access to Raspberry

andyzlen, 2018-11-03:
Hello framp,
with Raspibackup I'm facing following problem:
the second time now my backup of Raspberry USB device to my Unix NAS backup was not completed and now I can not access Raspberry any more, neither via monitor attached nor via SSH (but answers to ping and I can see green ACT_LED is flashing).
What happens in detail?
In Raspibackup log the last line documented is 
“RBK0044I: Backup der Bootpartition wird in /media/nas-backup/raspi/raspi-rsync-backup-20181103-110003/raspi-backup.img erstellt.”
After that raspberry is sending the log via email to my mail account – without any error message.
When I check my Raspberry afterwards -wondering why its sending me email in between the backup is running I can’t access any more.

Could you give any help how to proceed?

My environment:
Raspberry PI3 with Raspbian Jessie, USB boot device
NAS (ext3) attached via NFS / Hardlink
All devices attached via LAN
No change of OS versions (Raspberry, NAS) or Raspibackup before

Log which was sent via email:
```
--- RBK0009I: raspi: raspiBackup.sh V0.6.3 (d19c32d) um Sat Nov  3 11:00:03 CET 2018 gestartet.
--- RBK0128I: Logdatei ist /media/nas-backup/raspi/raspi-rsync-backup-20181103-110003/raspi-backup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0031I: PrÃ¼fe ob neue Version verfÃ¼gbar ist.
--- RBK0080I: ;-) Es gibt eine neue Version 0.6.4 von raspiBackup zum downloaden. Die momentan benutze Version ist 0.6.3 und es kann mit der Option -U die lokale Version aktualisiert werden. Besuche https://www.linux-tips-and-tricks.de/de/versionshistorie um die Ã„nderungen in der Version zu erfahren
--- RBK0114I: Besuche https://www.linux-tips-and-tricks.de/de/versionshistorie/ um die Ã„nderungen in der neuen Version kennenzulernen.
--- RBK0159I: PrÃ¼fe ob eine Beta Version verfÃ¼gbar ist.
--- RBK0168I: raspiBackup.sh Beta Version 0.6.4.1 ist verfÃ¼gbar. Hilfe beim Testen dieser Beta ist sehr willkommen..
--- RBK0151I: Backuppfad /media/nas-backup wird benutzt.
--- RBK0008I: Services werden gestoppt: 'systemctl stop openhab2.service && systemctl stop smbd.service && service nmbd stop && service smbd stop && service cron stop'.
--- RBK0081I: Backup vom Typ rsync wird in /media/nas-backup/raspi/raspi-rsync-backup-20181103-110003 erstellt.
--- RBK0036I: Partitionslayout wird gesichert.
--- RBK0044I: Backup der Bootpartition wird in /media/nas-backup/raspi/raspi-rsync-backup-20181103-110003/raspi-backup.img erstellt.
```
Regards,
Andi

framps, 2018-11-03:
Great you were able to restore your system from the backup :+1: 

Good idea to enable debuglog creation allthe time. Just pass the debug log to me when you face the same issue again.

framps, 2018-11-04:
I just noticed you use raspiBackup v0.6.3. There is a known issue of the error message logging of the backup tools. Either use the latest [0.6.4.1 beta](https://github.com/framps/raspiBackup/blob/beta/raspiBackup.sh) or enable the verbose mode `-v` in 0.6.3.

andyzlen, 2018-11-04:
Thanks for the tip - just changed to  0.6.4.1 beta.

framps, 2018-12-02:
Thank you very much @christianTF for your comment. 
@andyzlen : Please follow Christians advice and let us know whether this fixes the issue.

andyzlen, 2018-12-03:
@christianTF Thank you for your advice.
@framps Within the last 4 weeks the issue didn't appear. Because I'm using the older 3B model and not the B+ model, I would prefer to wait if the issue occurs one more time to send the debuglog to you.
Regards,
Andi

framps, 2018-12-03:
I suggest to close the issue for now and reopen it when you face the issue again :smiley:

andyzlen, 2018-12-04:
That's absolutely fine for me.
Regards,
Andi