framps/raspiBackup #86: Error message RBK0027E written even a partition is mounted

framps, 2018-12-01:
raspiBackup doesn't detect a mounted partition if the backup path is a symbolic link and writes message RBK0027E

framps, 2019-01-27:
I tried to reproduce the issue:
Given
```
mount | grep "^/dev"
/dev/sda1 on / type ext4 (rw,noatime,errors=remount-ro,data=ordered)
/dev/sdb1 on /disks/system type ext4 (rw,noatime,errors=remount-ro,stripe=4,data=ordered)
/dev/sdb5 on /disks/data type ext4 (rw,noatime,errors=remount-ro,data=ordered)
```
```
ls -la / | grep disks
lrwxrwxrwx   1 root root    13 Jan 27 18:42 b -> /disks/system
lrwxrwxrwx   1 root root    19 Mar 11  2018 backup -> /disks/data/backup/
drwxr-xr-x   5 root root  4096 Jan 19 19:31 disks
```
Both /b and /backup are identified as valid backup pathes.

Closing issue because it's not reproducible.

closed 2019-01-27 by framps

Raemsna, 2022-03-04:
I can second this. I have a hdd mounted via usb to raspi on path /mnt/Mediaserver_hdd
Due to file structure on the Mediaserver_hdd I have a symbolic link from /mnt/Backup to the given file structure e.g. /mnt/Mediaserver_hdd/dir1/dir2/Backup
When I now want to make a Backup to /mnt/Backup (which symbolic links to the hdd) I get the error

`pi@RaspiBuero:/usr/local/etc $ sudo raspiBackup.sh
--- RBK0009I: RaspiBuero: raspiBackup.sh V0.6.6 (c8d3058) Fr 4. Mär 09:01:29 CET 2022 gestartet.
--- RBK0151I: Backuppfad /mnt/Backup mit Partitionstyp Typ wird benutzt.
??? RBK0027E: Kein externes Gerät an /mnt/Backup verbunden. Die root Partition würde für das Backup benutzt werden.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0043I: Unvollständiges Backup /mnt/Backup//RaspiBuero/RaspiBuero-tgz-backup-20220304-090128 in wird gelöscht. Das kann etwas dauern. Bitte Geduld.
--- RBK0049I: Meldungen wurden in /home/pi/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert.
--- RBK0010I: RaspiBuero: raspiBackup.sh V0.6.6 (c8d3058) Fr 4. Mär 09:01:55 CET 2022 beendet mit Returncode 102.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.`

framps, 2022-03-04:
@Raemsna You can use the undocumented option `-c` which will turn off the mount check :wink: 

But then make sure you have the backup partition mounted all the time. Otherwise the backup will be stored on your root partition and may blow up your disk space and finally kill your system. In addition it's a bad idea to save the backup of the system on the root partition.