framps/raspiBackup #87: Operation not permitted (1)

EvilPingu, 2018-12-02:
Hi,

i want to use raspiBackup. If i'm executing raspiBackup i get the following errors:

rsync: mknod "/home/pi/shares/backup-pi/openhab/openhab-rsync-backup-20181202-133934/tmp/.X11-unix/X0" failed: Operation not permitted (1)
rsync: mknod "/home/pi/shares/backup-pi/openhab/openhab-rsync-backup-20181202-133934/tmp/.vnc-vncservice/vncserver-x11.CtrlComms" failed: Operation not permitted (1)
rsync: mknod "/home/pi/shares/backup-pi/openhab/openhab-rsync-backup-20181202-133934/tmp/.vncserver-license/0.440" failed: Operation not permitted (1)
rsync: mknod "/home/pi/shares/backup-pi/openhab/openhab-rsync-backup-20181202-133934/tmp/dhcpcd-pi/libdhcpcd-wpa-1284.0" failed: Operation not permitted (1)
rsync: mknod "/home/pi/shares/backup-pi/openhab/openhab-rsync-backup-20181202-133934/tmp/dhcpcd-pi/libdhcpcd-wpa-1284.1" failed: Operation not permitted (1)
rsync: mknod "/home/pi/shares/backup-pi/openhab/openhab-rsync-backup-20181202-133934/tmp/ssh-U8eJQrhVm1X1/agent.1138" failed: Operation not permitted (1)
rsync: mknod "/home/pi/shares/backup-pi/openhab/openhab-rsync-backup-20181202-133934/tmp/ssh-s5sJqX9QnJJd/agent.1291" failed: Operation not permitted (1)

I have no idea about this error. "/home/pi/shares/backup-pi" is the backup share. It's a cifs mounted device.

I'm wondering why something from the "/tmp" directory will be saved on the backup device. After backing up the backup will deleted.

Please help me

framps, 2018-12-02:
Hi EvilPingu,

you address multiple issues. Thank you for reporting them.

1) `rsync:mknod ... Operation not permitted (1)` tells you there is some authorization missing on your cifs mounted device. Please check your cifs configuration.
2) /tmp shouldn't be saved. I created an [issue for this](https://github.com/framps/raspiBackup/issues/88). 
3) It's possible to use rsync backup on a cifs mounted partition but keep in mind you will not save any disk space and backup time because hardlinks are not supported via cifs. Every backup run will save all files all the time.

EvilPingu, 2018-12-02:
> Hi EvilPingu,
>
> you address multiple issues. Thank you for reporting them.
>
>     1. `rsync:mknod ... Operation not permitted (1)` tells you there is some authorization missing on your cifs mounted [...]

Ok Thank you 👍

framps, 2018-12-02:
I fixed issue (2) right now. Please try the latest beta.

EvilPingu, 2018-12-13:
Hi,

i tested the issue. Now the temp-directory is not saved anymore... Great!

But...

there seems to be another problem. But i don't see the error. What does rc 109 mean?

/usr/include/netatalk/at.h
/usr/include/RTIMUMagCal.h
/usr/include/RTIMUCalDefs.h
/usr/include/ctype.h
/usr/include/mqueue.h
/usr/include/poll.h
20181213-214226: DBG                 -- Result 1
20181213-214226: DBG                 << tarBackup 1
12-13-2018 21:42:26 ??? RBK0021E: Backupprogramm des Typs tar beendete sich mit RC 1.
20181213-214226: MSG              12-13-2018 21:42:26 ??? RBK0021E: Backupprogramm des Typs tar beendete sich mit RC 1.
20181213-214226: DBG                 >> exitError 109
20181213-214226: DBG                 << exitError 109
12-13-2018 21:42:26 --- RBK0032I: Bitte warten bis aufgeräumt wurde.
20181213-214226: MSG              12-13-2018 21:42:26 --- RBK0032I: Bitte warten bis aufgeräumt wurde.
20181213-214227: DBG                 >> cleanupBackup
20181213-214227: DBG                 -- Got trap EXIT
20181213-214227: DBG                 -- rc: 109
Invocation parms: ' --systemstatus -l debug --timestamps -g -a : -o : .'
20181213-214227: DBG                    >> sendEMail
20181213-214227: DBG                    << sendEMail
20181213-214227: DBG                    >> cleanupBackupDirectory
20181213-214228: DBG                    -- BackupDir created: 1
20181213-214228: MSG                    12-13-2018 21:42:28 --- RBK0043I: Unvollständiges Backup /home/pi/shares/backup-pi/openhab/openhab/openhab-tar-backup-2$
20181213-214228: MSG                    12-13-2018 21:42:28 --- RBK0105I: Neues Backupverzeichnis /home/pi/shares/backup-pi/openhab/openhab/openhab-tar-backup-$
20181213-214229: DBG                    -- total 5931008
drwxr-xr-x 2 root root          0 Dec 13 21:32 .
drwxr-xr-x 2 root root          0 Dec 13 21:31 ..
-rwxr-xr-x 1 root root   46005248 Dec 13 21:32 openhab-backup.img
-rwxr-xr-x 1 root root        512 Dec 13 21:32 openhab-backup.mbr
-rwxr-xr-x 1 root root        195 Dec 13 21:32 openhab-backup.sfdisk
-rwxr-xr-x 1 root root 6023700480 Dec 13 21:42 openhab-tar-backup-20181213-213157.tar
-rwxr-xr-x 1 root root   12926104 Dec 13 21:42 raspiBackup.log
20181213-214229: MSG                    12-13-2018 21:42:29 --- RBK0026I: Logdatei wurde in /home/pi/raspiBackup.log gesichert.
20181213-214229: DBG                    << cleanupBackupDirectory
20181213-214229: MSG                 12-13-2018 21:42:29 ??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
20181213-214229: DBG                 << cleanupBackup
20181213-214229: DBG                 >> cleanupTempFiles
20181213-214229: DBG                 -- Removing mailfile /tmp/raspiBackup.maillog
20181213-214229: DBG                 << cleanupTempFiles
20181213-214229: DBG              -- Terminate now with rc 109
20181213-214229: MSG              12-13-2018 21:42:29 --- RBK0010I: openhab: raspiBackup.sh V0.6.4.1-beta (4a82b4f) Thu 13 Dec 21:42:29 CET 2018 beendet.

framps, 2018-12-13:
Thank you very much for verifying the fix.

RBK0021E is explained [here](https://www.linux-tips-and-tricks.de/en/rmessages#m0021). Actually a tar RC 1 usually means some file(s) changed during backup. You should check with option --systemstatus which files are open and stop the running service which writes the file which changes with option -o.

framps, 2018-12-30:
@EvilPingu Any update on the issue?