framps/raspiBackup #90: Create raspiBackup ui installer

framps, 2018-12-07:
Right now the current installer is just command line based. A convenient ui based installer is missing.

framps, 2018-12-07:
raspi-config uses whiptail which is an ASCII based ui. Simple but works as designed. Development of the ui installer is done on the ui_installer branch and considered beta.

christianTF, 2018-12-08:
From the LoxBerry unattended installation perspective, please keep an option to run without ui when install parameters are given.

From the users perspective, a ui creating a backup commandline or Default Config would be useful.

Lg, Christian

framps, 2018-12-08:
That's something I haven't had on my radar :cry: but makes absolutely sense.

Let's elaborate on this a bit more: As far as I understand you want to have an additional option (let's use -U for `unattended` which enables raspiBackup unattended installation mode of the ui_installer and uses the default configuration defined in the configuration [template downloaded during installation](https://www.linux-tips-and-tricks.de/downloads/raspibackup-en-conf/download).

Following configurations can be changed via the ui_installer afterwards:
* backuptype 
* backupmode
* keepbackups 
* zip 
* message level
* message language
* cron backup weekday (Default So) 
* cron backup time (Default 5 am)

Other options have to be modified via an editor in the config file.

. In addition another option -E will enable the unattended installation of the sample extensions.

Sounds this reasonable for you?

I frankly don't get `From the users perspective, a ui creating a backup commandline`.

christianTF, 2018-12-08:
I haven’t investigated in a manual install, as I currently use the installer.
I also can copy the required files for the backup myself, including a default config, then you don’t need to create extra parameters for unattended installations.

From the users perspective it may be useful to create ad hoc backups of different types or different settings, that currently need to investigate in all the parameters. It would be helpful to have some kind of wizard to create such a command line.

For myself, my demands are fulfilled in the current version.

framps, 2019-03-05:
UI installer is now no longer beta ac89a12d6c6736c5e7f72065d633b92bb4e85898

closed 2019-03-05 by framps