framps/raspiBackup #92: printf error in line 939

christianTF, 2018-12-29:
Hi Framp,

in the Beta (see log below) I get a printf error in line 939.

```
INFO: sudo /usr/local/bin/raspiBackup.sh -N "fstab disk temp" -o "systemctl stop mariadb" -a "systemctl start mariadb" -k 5 -t ddz -L current -l debug -m detailed -v /opt/loxberry/system/storage/smb/homeserver/Raspberry-Backups 
Script started on Sat Dec 29 09:54:03 2018
--- RBK0009I: papi: raspiBackup.sh V0.6.4.1-beta (007a287) started at Sat Dec 29 09:54:03 CET 2018. 
!!! RBK0165W: =========> NOTE <========= 
!!! RBK0165W: This is a betaversion and should not be used in production. 
!!! RBK0165W: =========> NOTE <========= 
--- RBK0128I: Using logfile /opt/loxberry/log/plugins/lbbackup/raspiBackup.log. 
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf. 
--- RBK0151I: Using backuppath /media/smb/homeserver/Raspberry-Backups. 
--- RBK0008I: Stopping services: 'systemctl stop mariadb'. 
--- RBK0081I: Creating backup of type ddz in /media/smb/homeserver/Raspberry-Backups/papi/papi-ddz-backup-20181229-095403. 
/usr/local/bin/raspiBackup.sh: line 939: printf: --: invalid option 
printf: usage: printf [-v var] format [arguments] 
--- RBK1005I: Copy %s to %s 
total 1 
drwxrwxr-x 2 loxberry loxberry 0 Dec 29 2018 . 
drwxrwxr-x 2 loxberry loxberry 0 Dec 29 2018 .. 
-rw-rw-r-- 1 loxberry loxberry 628 Dec 29 2018 fstab 
--- RBK0085I: Backup of type ddz started. Please be patient. 
```

https://github.com/framps/raspiBackup/blob/b6666b77070c58066607677133b383de3562a581/raspiBackup.sh#L939

I'm not sure, but is it a follow-up error because of a problem with language detection, or is a language string missing?

Thanks,
Christian

framps, 2018-12-29:
Hello Christian,

thank you very much for reporting the issue. root cause is you use the fstab_ready extension version from 0.6.4. In 0.6.4.1 the message format has changed and the messages were updated accordingly. Please use the extension from the beta branch (See [diff here](https://github.com/framps/raspiBackup/commit/80e93a0c8621ce1b064615cba3fbe64b3730a263#diff-d0d6a035dd5f0d29ac809a4017146d41) ) and the issue will be gone.

You raised an interesting issue for 0.6.4.1 I wasn't aware about: When publishing 0.6.4.1 I also have to alert everybody using extensions to upgrade them also.

framps, 2019-01-04:
I just added some code in 0.6.4.1 which converts the old message format used in extensions into the new format. That way raspiBackup 0.6.4.1 beta can handle the old message format also and there is no need to upgrade the extensions together with raspiBackup to 0.6.4.1.

@christianTF Would be great if you verify the new message compatibility code when you have some spare time :smile:

closed 2019-01-12 by framps