framps/raspiBackup #94: RaspiBackup terminated with RC 23 – xattrs with rsync

andyzlen, 2019-01-05:
Hello framps,
after upgrade of some SW on my raspberry (e.g. openhab) raspibackup terminated with error during backup procedure, please see below. Before this upgrade the backup was successful with raspibackup version V0.6.4.1-beta (which was not changed).

Error message in raspiBackup.log:
`rsync: rsync_xal_set: lsetxattr(""/media/nas-backup/raspi/raspi-rsync-backup-20190105-080004/usr/bin/.systemd-detect-virt.ux6PPI"","security.capability") failed: Operation not supported (95)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1183) [sender=3.1.1]
…
??? RBK0021E: Backupprogramm des Typs rsync beendete sich mit RC 23.
`
I suppose my issue is related to this #66 Check whether rsync partition supports xattrs.
That’s why I comment out the parameter DEFAULT_RSYNC_BACKUP_OPTIONS="-aHAXx" in my (quite old) raspiBackup.conf which do not exist in that file version available on your website.

Could you confirm if this approach was correct?

My environment:
Raspberry PI3 with Raspbian Jessie (kernel Linux raspi 4.9.35-v7+), USB boot device
raspibackup version V0.6.4.1-beta
Backup on NAS (ext3) attached via NFS / Hardlink
All devices attached via LAN

Regards,
Andi

andyzlen, 2019-01-05:
Backup with adapted raspiBackup.conf was successful. But I'm not sure if ignoring extended file attributes could have side effects on a restored backup.
Regards,
Andi

framps, 2019-01-05:
Hello andyzlen,

`DEFAULT_RSYNC_BACKUP_OPTIONS="-aHAx"` is now the default in the 0.6.4.1 beta and still can be overwritten in the config. But I decided to hide this because these options are sensible and shouldn't be modified by users.

This means raspiBackup will not save xattrs with rsync any more. Whether this is an issue depends on the program which uses these xattrs because a restored image will not have them. Usually it's something related to SELinux.

You can search for the files using xattrs with
`find / -xattr -exec xattr -v {} \;`
which may help to isolate the program using the xattrs.

I see following solutions for you:

1) Use tar backup which will save xattrs
2) Use rsync backup and locate the program using the attributes and find out whether it's an important program or the missing attributes can be ignored (that's what raspiBackup users usually do - you are not the first one hitting this issue :smile: )
3) Use rsync backup with a local attached ext3/4 partition

Hope this helps

andyzlen, 2019-01-05:
Thank you @framps for the quick answer and your suggestions. I decided to take option 2. 
Only Raspbian Jessie's find command doesn't support "-xattr".
Regards,
Andi

framps, 2019-01-05:
> Only Raspbian Jessie's find command doesn't support "-xattr".

Sorry. Try 
`find / \( -path /proc -o -path /dev -o -path /proc -o -path /sys -o -path /tmp -o -path /run -o -path /mnt -o -path /media -o -path /lost+found -o -path /export -o -path /home/ftp -o -path /srv -o -path /sharedfolders \) -prune -o -name '*' -exec getfattr -h -d -m "^" --absolute-names {} \; 2> /dev/null | more`

framps, 2019-01-06:
Great we found a solution for you. Please close the issue if you're fine now :smile:

closed 2019-01-06 by andyzlen