framps/raspiBackup #97: Email notification only on failure

whowantsmybigdata, 2019-01-15:
Is it or wouldn't it be possible to only notify me via email when the backup fails?! As i do a lot of unattended rsync backups which almost always succeed getting dozens of confirmation mails is quite a waste... Keep up the good work and thanks a lot!!

whowantsmybigdata, 2019-01-15:
Thanks a lot! Answered via email.
By the way: is there a way to donate without paypal?! best would be bitcoins?

framps, 2019-01-15:
NP. There are a lot of options and you're not the only one who missed an option :wink: 

There was already one user who didn't want to use paypal either. He just sent me a letter with some bucks. If that's fine with you just [lookup my eMail](https://www.linux-tips-and-tricks.de/en/contact) and ask for my address :smile: 

Please close this issue if the option works fine for you.

whowantsmybigdata, 2019-01-16:
Works flawlessly... 
Hm, strange options to donate, on one hand paypal on the other hand a letter?! haha,
I'll think about which method I dislike more and contact you in case of the letter! 
Thanks again for such a useful, nice working program!

framps, 2019-01-16:
if you have an link to a howto handy which allows me to setup my BC wallet on Linux in 5 minutes I will give it a try :-)

whowantsmybigdata, 2019-01-17:
so i use [electrum](https://electrum.org/#download):

install is quite straightforwards and more or less self-explainable but theres a [detailed doc](electrum.readthedocs.io) if you need extra security or general help.

whowantsmybigdata, 2020-02-24:
Hi,
out of nowhere (without changing anything in raspibackup-conf or with the system at all and without upgrading) I got myself a very similar problem... 

Using version 0.6.4.3 (working normally before) aswell as 0.6.4.5 Beta:

If I trigger a manual backup via <code>/usr/local/bin/raspiBackup.sh</code> everything works fine and I would only get an email-notification in case of a failure.
If cron triggers the backup via <code>/usr/local/bin/raspiBackup.sh > /dev/null 2>&1</code> I get an email-confirmation on failure AND even if everything worked, which it does.
I also tried adding <code> DEFAULT_MAIL_ON_ERROR_ONLY=1 </code> with no success.

[raspiBackup-cron.log](https://github.com/framps/raspiBackup/files/4246893/raspiBackup-cron.log)
[raspiBackup-manually.log](https://github.com/framps/raspiBackup/files/4246894/raspiBackup-manually.log)

[config.log](https://github.com/framps/raspiBackup/files/4246897/config.log)

Appreciate your help!

framps, 2020-02-25:
> Still strange of the raspibackup-script to send an email ignoring the config just because the output is discarded by cron, isn't it?

Just to make sure I get you correctly:
1. You set `DEFAULT_MAIL_ON_ERROR_ONLY=1` in config
2. If you use 0.6.4.3 or beta you get a success email notification if you use `>/dev/null 2>&1` in crontab.
3. If you use 0.6.4.3 from the commandline you don't get a success eMail
4. If you don't use `>/dev/null 2>&1` you don't get a success email notification 
5. In order to suppress the cron eMail you set `MAILTO:"" `

There is some code which differentiates whether raspiBackup is called from a terminal or in the background. It's just used to redirect all &1 and &2 output into the log file instead to the console and shouldn't be the root cause for your issue.

I frankly don't think it's a good idea to suppress cron eMails because that's the way you get notified if something went wrong in a cron job. what's the contents of the eMail do you get if you don't have  `MAILTO:"" ` in your cronjob?

whowantsmybigdata, 2020-02-25:
> Just to make sure I get you correctly:
>
>     1. You set `DEFAULT_MAIL_ON_ERROR_ONLY=1` in config
>
>     2. If you use 0.6.4.3 or beta you get a success email notification if you use `>/dev/null 2>&1` [...]

All correct! I only set <code>MAILTO=""</code> for the particular raspibackup.sh-command and thought in case of a  failure raspiback will send an email anyway... 
The mail from cron includes the output of the stop and start services just like the terminal output when triggered manually.

framps, 2020-02-25:
I tried to reproduce your issue but I fail. But I noticed I get an eMail from cron (not from raspiBackup) when there was some output written by raspiBackup. raspiBackup usually writes everything to the log when running in background. Some services started and/or stopped with option -a and -o don't use the usual usage of stdout and sterr and thus raspiBackup creates some output which will be sent in an eMail by cron. You can suppress these eMails by redirection everything to /dev/null what you did. But I didn't get an eMail when I had a successfully backup run. 

I noticed I get an eMail from cron when a debug statement in raspiBackup created some error message. But that's an eMail sent by cron - not by raspiBackup. To sum it up: I'm not able to reproduce your issue. `DEFAULT_MAIL_ON_ERROR_ONLY` works for me fine if I redirect stdout and stderr into /dev/null or not. 

Do you know [this page](http://www.auxnet.org/index.php/the-news/228-stop-cron-daemon-from-sending-email-for-each-job) which summarizes how to get rid of any cron eMails?

whowantsmybigdata, 2020-02-25:
Yes, I guess this is the problem with my setup. Starting/Stopping Seafile is causing extra output…
Nevertheless its strange you can put the output to /dev/null without getting an email from raspibackup.
Anyway, my workaround works for now. 
Thanks for the website-hint will look it up if I get the time… and a big thanks for your nice program (saved my ass one time since I opened this issue) and for your fast support!
Greets

closed 2020-02-25 by whowantsmybigdata