framps/raspiBackup #100: Release naming and upgrade rule

cgail914, 2019-01-18:
Hi,
no big deal but for the sake of being clean.

I have 2 systems:
one was running 0.6.3, I did a -U execution, and it just got upgraded to 0.6.4.1a
one was running 0.6.4.1, I did a -U execution, and it stayed on 0.6.4.1.

I can't find 0.6.4.1a here on github but finds the release on the webpage of raspibackup. Is it a side release ? if yes, why then -U installs a non-official/stable release ?
if it's a true release, then can we make it official on GitHub and make sure that ANY previous release incl. 0.6.4.1 gets upgraded to the latest with a -U.

Sorry if I overlooked something

Cheers

ps. would be good if the id linked to a release (exemple V0.6.4.1 (fc71f93) <-- this is the id), is the same id shown on github, so that we can easily match them between what's installed and what's on github.

framps, 2019-01-18:
Hello @cgail914 ,

> no big deal but for the sake of being clean.
Your comments are valid. Thank you very much for creating this issue.

> one was running 0.6.3, I did a -U execution, and it just got upgraded to 0.6.4.1a
> one was running 0.6.4.1, I did a -U execution, and it stayed on 0.6.4.1.

That's a bug. I`created https://github.com/framps/raspiBackup/issues/101 for this.

> I can't find 0.6.4.1a here on github but finds the release on the webpage of raspibackup. Is it a side release ? if yes, why then -U installs a non-official/stable release ?

I use the suffix a,b,c for minor fixes which I don't want to be notified about in a message by raspiBackup. It's supposed to notify only official releases as 0.6.4.1 but not 0.6.4.1a. But upgrades with -U should upgrade to the latest release (That's the bug you found).

> ps. would be good if the id linked to a release (exemple V0.6.4.1 (fc71f93) <-- this is the id), is the same id shown on github, so that we can easily match them between what's installed and what's on github.

You rub salt into my wounds ... I know I'm weak on this. It's because I decided to spend my limited spare time on bug fixes and enhancements and not to setup a stable deployment environment which has to be thoroughly be tested. 
Right now I deploy a new release manually and sometimes I miss an important step - for example sync the commit sha :cry: 

I created https://github.com/framps/raspiBackup/issues/102 for this.

cgail914, 2019-01-19:
Hi @framps ,
thanks for the feedback. And thanks for the software, it does make a difference !
Best

closed 2019-01-19 by cgail914

framps, 2019-01-19:
@cgail914 
FYI: I just added a workaround for the update issue you reported in https://github.com/framps/raspiBackup/issues/101