framps/raspiBackup #102: Create a stable and consistent publish procedure (makefile)

framps, 2019-01-18:
> ps. would be good if the id linked to a release (exemple V0.6.4.1 (fc71f93) <-- this is the id), is the same id shown on github, so that we can easily match them between what's installed and what's on github.

Right now every release is published manually which may cause some inconsistencies. A makefile with appropriate targets should solve this issue.

framps, 2019-03-05:
Root cause for the discrepancy between commit sha used in the production code and the commit sha in this repo is that deployment of the code into production is done from my local repository and the local commit sha is used. This commit sha does not exist in this git repository and thus there is no way to correlate the deployed code with any commits here.

cgail914, 2019-03-09:
No, it's not. I can live with it. thx.

framps, 2019-03-09:
Great. I know that's not the common way to publish code on github. But I have some strong reasons I do it this way. 
There is a workaround: Just search for the sha of your current raspiBackup version in this repository and you will find the code for this commit sha :smiley: 

Ok to close this issue?

cgail914, 2019-03-10:
Sure.

closed 2019-03-10 by framps