framps/raspiBackup #103: tar backup exclude dir

a323870, 2019-01-19:
Tar img non include dir /proc /lost+found /sys /dev /tmp /boot /run

I solve append /*  at the end of --exclude

	cmd="tar \
		$TAR_BACKUP_OPTIONS \
		$TAR_BACKUP_ADDITIONAL_OPTIONS \
		${zip} \
		${verbose} \
		-f $target \
		--warning=no-xdev \
		--numeric-owner \
		--exclude=\"$BACKUPPATH_PARAMETER/*\" \
		--exclude=\"$source/$log_file\" \
		--exclude='.gvfs' \
		--exclude=$devroot/proc/* \
		--exclude=$devroot/lost+found/* \
		--exclude=$devroot/sys/* \
		--exclude=$devroot/dev/* \
		--exclude=$devroot/tmp/* \
		--exclude=$devroot/boot/* \
		--exclude=$devroot/run/* \
		$EXCLUDE_LIST \
		$source"

framps, 2019-01-19:
Hello @a323870 ,

you are right. Without /* the contents of these directories is not saved. But all these directories are recreated during Linux boot so there is no need to save them. 

What's the reason you want to save these directories? If you have a strong argument for this I will add an additional option to raspiBackup in next release for this.

a323870, 2019-01-19:
i test  restore phase in a new sd card but at first boot raspberry don't
create this dir.
Manually create on new SD and boot is ok.

Tank.

a323870, 2019-01-19:
> Do you use the USB boot mode of your Raspberry and boot without any SD
> card?

No only sd card.
Boot start, but end in error not found /dev /sys ecc....
Create folder manually mounted sd in another raspy.
Reinsert sd and reboot first raspy its ok.

framps, 2019-01-19:
Which OS do you use? Please provide the output of
`cat /etc/os-release`

a323870, 2019-01-19:
a323870@raspberrypi ~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
a323870@raspberrypi ~ $

Standard raspbian.
The problem is that this folder aren't create at first boot.
I saved only the folder in tar image.

Bye.

framps, 2019-01-19:
Thank you very much for this information. I don't use tar backup on my Raspis but it's tested in the raspiBackup regression test and I'm wondering why this issue doesn't pop up there. 

I have to investigate on this further now.

framps, 2019-01-19:
> If it can bu usful 
:+1:

framps, 2019-01-19:
Thank you very much you reported this valid issue. :+1:

I just fixed it and decided to make it a new release 0.6.4.2 so everybody using raspiBackup will get notified to upgrade.

Just execute `raspiBackup.sh -U` to get the latest release.

a323870, 2019-01-19:
I appreciated your work. I use this script to save my three raspberry
every month via cron.

Best Regards.

framps, 2019-01-20:
Fixed in 0.6.4.2

closed 2019-01-20 by framps

framps, 2019-01-20:
@a323870 I appreciate you created this issue and helped to find the root cause :+1: