framps/raspiBackup #107: tar backup fails on raspiBackup 6.4.2

whowantsmybigdata, 2019-01-29:
Hi,
unfortunately my tar-backups started to fail a few days ago and as I weren't able to fix it myself maybe somebody here could help. 
Failure started on version 6.4.1 but is occuring after upgrade to 6.4.2 aswell.
My guess is it has something to do with the socket for ccnet (as part of seafile)?!

<details>
  <summary>Anyway, this is the log:</summary>
  <p>
<pre>
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
20190129-031313: MSG     --- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
--- RBK0151I: Backuppfad /media/Server-Backup wird benutzt.
20190129-031321: MSG        --- RBK0151I: Backuppfad /media/Server-Backup wird benutzt.

Stopping seahub ...
Done.

Stopping seafile server ...
Done.
--- RBK0036I: Partitionslayout wird gesichert.
20190129-031333: MSG                 --- RBK0036I: Partitionslayout wird gesichert.
1161+0 records in
1161+0 records out
1217396736 bytes (1.2 GB, 1.1 GiB) copied, 38.0641 s, 32.0 MB/s
1+0 records in
1+0 records out
512 bytes copied, 0.00149166 s, 343 kB/s
--- RBK0158I: tar Backup "/media/Server-Backup/Raspbian/Raspbian-tar-backup-20190129-031310/Raspbian-tar-backup-20190129-031310.tar" wird erstellt.
20190129-031412: MSG              --- RBK0158I: tar Backup "/media/Server-Backup/Raspbian/Raspbian-tar-backup-20190129-031310/Raspbian-tar-backup-20190129-031310.tar" wird erstellt.
--- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld.
20190129-031412: MSG              --- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld.
tar: Removing leading `/' from member names
tar: Removing leading `/' from hard link targets
tar: /home/seafile/ccnet/ccnet.sock: socket ignored
tar: /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190129-031310/Raspbian-tar-backup-20190129-031310.tar: Wrote only 8192 of 10240 bytes
tar: Error is not recoverable: exiting now
??? RBK0021E: Backupprogramm des Typs tar beendete sich mit RC 2.
20190129-043852: MSG           ??? RBK0021E: Backupprogramm des Typs tar beendete sich mit RC 2.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
20190129-043852: MSG           --- RBK0032I: Bitte warten bis aufgeräumt wurde.

Seahub is not running
Done.

seafile server not running yet
Done.

[01/29/19 04:39:12] ../common/session.c(132): using config file /home/seafile/conf/ccnet.conf
Starting seafile server, please wait ...
Seafile server started

Done.

Starting seahub at port 8000 ...

Seahub is started

Done.

Invocation parms: ''
--- RBK0043I: Unvollständiges Backup /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190129-031310 in wird gelöscht. Das wird etwas dauern. Bitte Geduld.
20190129-043926: MSG                 --- RBK0043I: Unvollständiges Backup /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190129-031310 in wird gelöscht. Das wird etwas dauern.$
--- RBK0105I: Neues Backupverzeichnis /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190129-031310 wird gelöscht.
20190129-043926: MSG                 --- RBK0105I: Neues Backupverzeichnis /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190129-031310 wird gelöscht.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
20190129-043927: MSG              ??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: Raspbian: raspiBackup.sh V0.6.4.2 (67129a2) Tue Jan 29 04:39:27 CET 2019 beendet.
20190129-043927: MSG           --- RBK0010I: Raspbian: raspiBackup.sh V0.6.4.2 (67129a2) Tue Jan 29 04:39:27 CET 2019 beendet.
</pre>
</p>
</details>

<details>
<summary> and this my config </summary>
<p>
<pre>
#
# Default configuration file for raspiBackup.sh
#
# (C) 2013,2016 framp at linux-tips-and-tricks dot de
#
# Details for the parameters see
# http://www.linux-tips-and-tricks.de/de/raspibackup/#parameter
#
# $Date: 2016-04-14 23:46:02 +0200$
# $Sha1: 04875c3baad59196d9b4c48bfb81b48764e3b03a$

# path to store the backupfile
DEFAULT_BACKUPPATH="/media/Server-Backup"

# how many backups to keep
DEFAULT_KEEPBACKUPS="1"

# type of backup: dd, tar or rsync
DEFAULT_BACKUPTYPE="tar"

# zip tar or dd backup (0 = false, 1 = true)
DEFAULT_ZIP_BACKUP="0"

# commands to stop services before backup separated by &
DEFAULT_STOPSERVICES="systemctl restart monit && systemctl stop nfs-kernel-server && systemctl stop cron && systemctl stop nginx && /home/user/seafile-stop.sh && systemctl stop netatalk && systemctl stop ejabberd && systemctl stop monit"

# commands to start services after backup separated by &
DEFAULT_STARTSERVICES="systemctl start monit && systemctl start ejabberd && systemctl start cron && systemctl start nginx && systemctl start nfs-kernel-server && /home/user/seafile-restart.sh && systemctl start netatalk"

# email to send completion status
DEFAULT_EMAIL=„eMail@domain“ 
DEFAULT_MAIL_ON_ERROR_ONLY="1"

# Additional parameters for email program (optional)
DEFAULT_EMAIL_PARMS="" 

# log level  (0 = none, 1 = debug)
DEFAULT_LOG_LEVEL=0

# log output ( 0 = syslog, 1 = /var/log, 2 = backuppath)
DEFAULT_LOG_OUTPUT=1

# msg level (0 = minimal, 1 = detailed)
DEFAULT_MSG_LEVEL="0"

# mailprogram
DEFAULT_MAIL_PROGRAM="ssmtp"

# restore device
DEFAULT_RESTORE_DEVICE=""

# default append log (0 = false, 1 = true)
DEFAULT_APPEND_LOG=0

# default verbose log (0 = false, 1 = true)
DEFAULT_VERBOSE=0 

# skip check for remote mount of backup path (0 = false, 1 = true)
DEFAULT_SKIPLOCALCHECK=0 

# blocksize used for dd 
DEFAULT_DD_BLOCKSIZE=1MB

# addition parms used for dd 
DEFAULT_DD_PARMS=""

# exclude list
#DEFAULT_EXCLUDE_LIST=""

# notify in email if there is an updated script version available  (0 = false, 1 = true)
DEFAULT_NOTIFY_UPDATE=1 

# extensions to call
DEFAULT_EXTENSIONS=""

# partition based backup  (0 = false, 1 = true)
DEFAULT_PARTITIONBASED_BACKUP="0"

# partition numbers of partitions to save in partitionbased backup mode
DEFAULT_PARTITIONS_TO_BACKUP="*"

# language (DE or EN)
DEFAULT_LANGUAGE="DE"

# hosts which will get the updated backup script with parm -y - non pwd access with keys has to be enabled
# Example: "root@raspifix root@raspberrypi root@raspiap root@raspibmc pi@seafile"
DEFAULT_DEPLOYMENT_HOSTS=""

# Use with care !
DEFAULT_YES_NO_RESTORE_DEVICE="loop"

# with dd backup save only space used by partitions
DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0

# Use hardlinks for partitionbootfiles
DEFAULT_LINK_BOOTPARTITIONFILES=0

# !!! Change these options only if you know what you are doing !!! 
# Define the options used by tar and rsync to backup the data
DEFAULT_RSYNC_BACKUP_OPTIONS="-aHAXx"
DEFAULT_RSYNC_BACKUP_ADDITIONAL_OPTIONS=""
DEFAULT_TAR_BACKUP_OPTIONS="-cpi"
DEFAULT_TAR_BACKUP_ADDITIONAL_OPTIONS=""

# GENERATED - DO NOT DELETE 
UUID=6c4984e4-d79b-41eb-854e-605d8a9eef67
</pre>
</p>
</details>

(I have no idea why it puts the lines commented out by "#" into bold in here, sorry)

The selfmade start/restart scripts for seafile are working perfectly fine when running them solo. 

By the way: running a rsync-Backup with the exact same config (both hardware and software) is working flawlessly but I like to do a seperate tar-backup from time to time. 

Thanks for your help.

framps, 2019-01-29:
Hello whowantsmybi,

`DEFAULT_TAR_BACKUP_OPTIONS="-cpi"` seems to be the root cause for your issue. You didn't post the debug log but I expect you have mounted some external partition which causes your issue.

Please see #106 and #107 and update you config file accordingly.

Sorry for the inconvenience.

whowantsmybigdata, 2019-02-01:
Hi,
thanks for the answer. I let Raspbian run with some different configs during the last nights but unfortunately tar-backup fails the same way as before...
I didnt post the debug log because of security concerns as there is all system-files included. 
Strangly as well: The startup script for seafile is working flawlessly when triggered manually as root or via rootuser-cron. 
<details><summary>Also unexplanable to me: although I set "append log=1", I only get an email via cron saying:</summary>
<p><pre>tar: Removing leading `/' from member names
tar: Removing leading `/' from hard link targets
tar: /var/log/raspiBackup/Raspbian.log: file changed as we read it
tar: /home/seafile/ccnet/ccnet.sock: socket ignored

Error: [Errno 13] Permission denied: '/root'</pre></p></details>

<details>
<summary>Here is the debug log without the list of system-files</summary>
<p><pre>
20190201-030005: DBG  -- LOG_OUTPUT: 1
20190201-030005: DBG  -- Using logfile /var/log/raspiBackup/Raspbian.log
20190201-030005: DBG  -- --- RBK0009I: Raspbian: raspiBackup.sh V0.6.4.2 (67129a2) Fri Feb  1 03:00:05 CET 2019 gestartet.
20190201-030005: DBG  -- BACKUPTARGET_DIR: /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190201-030001
20190201-030005: DBG  -- BACKUPTARGET_FILE: /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190201-030001/Raspbian-tar-backup-20190201-030001.tar
20190201-030005: DBG     >> logOptions
20190201-030005: DBG     -- Linux Raspbian 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
20190201-030005: DBG     -- Options: 
20190201-030005: DBG     << logOptions
20190201-030005: DBG  -- APPEND_LOG=1
20190201-030005: DBG  -- APPEND_LOG_OPTION=-a
20190201-030005: DBG  -- BACKUPPATH=/media/Server-Backup
20190201-030005: DBG  -- BACKUPTYPE=tar
20190201-030005: DBG  -- CHECK_FOR_BAD_BLOCKS=1
20190201-030005: DBG  -- CONFIG_FILE=
20190201-030005: DBG  -- DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=1
20190201-030005: DBG  -- DD_BLOCKSIZE=1MB
20190201-030005: DBG  -- DD_PARMS=
20190201-030005: DBG  -- DEFAULT_DEPLOYMENT_HOSTS=
20190201-030005: DBG  -- DEFAULT_YES_NO_RESTORE_DEVICE=
20190201-030005: DBG  -- EMAIL=[myemailadress]
20190201-030005: DBG  -- EMAIL_PARMS=
20190201-030005: DBG  -- EXCLUDE_LIST=
20190201-030005: DBG  -- EXTENSIONS=
20190201-030005: DBG  -- FAKE=0
20190201-030005: DBG  -- HANDLE_DEPRECATED=1
20190201-030005: DBG  -- KEEPBACKUPS=1
20190201-030005: DBG  -- LANGUAGE=DE
20190201-030005: DBG  -- LINK_BOOTPARTITIONFILES=0
20190201-030005: DBG  -- LOG_LEVEL=1
20190201-030005: DBG  -- LOG_OUTPUT=1
20190201-030005: DBG  -- MAIL_ON_ERROR_ONLY=1
20190201-030005: DBG  -- MAIL_PROGRAM=ssmtp
20190201-030005: DBG  -- MSG_LEVEL=1
20190201-030005: DBG  -- NOTIFY_UPDATE=1
20190201-030005: DBG  -- PARTITIONBASED_BACKUP=0
20190201-030005: DBG  -- PARTITIONS_TO_BACKUP=*
20190201-030005: DBG  -- RESIZE_ROOTFS=1
20190201-030005: DBG  -- RESTORE_DEVICE=
20190201-030005: DBG  -- RESTORE_REMINDER_INTERVAL=6
20190201-030005: DBG  -- RESTORE_REMINDER_REPEAT=3
20190201-030005: DBG  -- ROOT_PARTITION=
20190201-030005: DBG  -- RSYNC_BACKUP_ADDITIONAL_OPTIONS=
20190201-030005: DBG  -- RSYNC_BACKUP_OPTIONS=-aHAx
20190201-030005: DBG  -- RSYNC_IGNORE_ERRORS=
20190201-030005: DBG  -- SENDER_EMAIL=
20190201-030005: DBG  -- SKIPLOCALCHECK=0
20190201-030006: DBG  -- STARTSERVICES=systemctl start monit && systemctl start ejabberd && systemctl start cron && systemctl start nginx && systemctl start nfs-kernel-server && /home/fabi/seafile-restart.sh && systemctl start netatalk
20190201-030006: DBG  -- STOPSERVICES=systemctl restart monit && systemctl stop nfs-kernel-server && systemctl stop cron && systemctl stop nginx && /home/fabi/seafile-stop.sh && systemctl stop netatalk && systemctl stop ejabberd && systemctl stop monit
20190201-030006: DBG  -- SYSTEMSTATUS=1
20190201-030006: DBG  -- TAR_BACKUP_ADDITIONAL_OPTIONS=
20190201-030006: DBG  -- TAR_BACKUP_OPTIONS=-cpi --one-file-system
20190201-030006: DBG  -- TAR_BOOT_PARTITION_ENABLED=0
20190201-030006: DBG  -- TAR_IGNORE_ERRORS=
20190201-030006: DBG  -- TAR_RESTORE_ADDITIONAL_OPTIONS=
[... 545 lines cut ...]
Seahub is not running
Done.

seafile server not running yet
Done.

[02/01/19 03:24:32] ../common/session.c(132): using config file /home/seafile/conf/ccnet.conf
Starting seafile server, please wait ...
Seafile server started

Done.

Starting seahub at port 8000 ...

Error: [Errno 13] Permission denied: '/root'
[33mError:Seahub failed to start.[m
Please try to run "./seahub.sh start" again
20190201-032443: DBG                 << executeShellCommand: 1
02-01-2019 03:24:43 ??? RBK0047E: Ein Fehler trat beim Starten von Services auf. RC 1.
20190201-032443: MSG              02-01-2019 03:24:43 ??? RBK0047E: Ein Fehler trat beim Starten von Services auf. RC 1.
20190201-032443: DBG                 >> exitError 106
20190201-032443: DBG                 << exitError 106
</pre></p></details>

Thanks again...

framps, 2019-02-01:
@whowantsmybigdata I deleted most of the log you posted because I considered some information in there as sensitive. But before I did this I saved the log on my local system :smiley: 

You reported now 3 issues:
1. append log=1 does not work
I will check what's going on there. But I think that's not the most important issue
2. You get a cron mail with some output. 
That's a known issue. See #108 for the fix and workaround.
3. Backup fails. 
In the log I found
```
Starting seahub at port 8000 ...
Error: [Errno 13] Permission denied: '/root'
�[33mError:Seahub failed to start.�[m
Please try to run "./seahub.sh start" again
20190201-032443: DBG                 << executeShellCommand: 1
02-01-2019 03:24:43 ??? RBK0047E: Ein Fehler trat beim Starten von Services auf. RC 1.
```
raspiBackup terminates unsuccessfully because during start of seahub a permission denied error happens. You have to find out why this error happens and fix it. Then the backup should succeed again.

whowantsmybigdata, 2019-02-02:
Alright... I thought mountpoints etc. wouldnt matter but thanks a lot anyway!
I didn't think a messed up startup script would fail the whole backup. Isn' it technically possible to just give a warning but still finishing the backup? 
Anyhow I figured out the problem with those messed up permissions in the startup script of seafile and as far as i know I got rid of it.

This beeing said, the backup still failed last night (as this is a production server with ejabberd running for quite some users I only do backups during the night using cron) with an eMail coming from cron saying:
<code>/usr/local/bin/raspiBackup.sh: line 2387: mpack: command not found</code>

That might be the cause for issue number 1 regarding to your previous post?! I installed mpack and will try again next night...

For you info, I did upgrade the backupscript as mentioned in #108 , so this mpack-failure happend with the newest version of the script.

<details><summary>But in the debug log the tar-Backup also fails with RC1</summary>
<p><pre> 20190202-032624: DBG              -- Result 1
20190202-032624: DBG              << tarBackup 1
02-02-2019 03:26:24 ??? RBK0021E: Backupprogramm des Typs tar beendete sich mit RC 1.
20190202-032624: MSG           02-02-2019 03:26:24 ??? RBK0021E: Backupprogramm des Typs tar beendete sich mit RC 1.
20190202-032624: DBG              >> exitError 109
20190202-032624: DBG              << exitError 109
02-02-2019 03:26:24 --- RBK0032I: Bitte warten bis aufgeräumt wurde.
20190202-032624: MSG           02-02-2019 03:26:24 --- RBK0032I: Bitte warten bis aufgeräumt wurde.
20190202-032624: DBG              >> startServices
20190202-032624: DBG                 >> logSystemStatus
20190202-032633: DBG                 -- service --status-all

[list of services deleted]

tee       5374             root    3w   REG  179,2  8842577 3539387 /var/log/raspiBackup/Raspbian.log
20190202-032634: DBG                 << logSystemStatus
02-02-2019 03:26:34 --- RBK0007I: Services werden gestartet: 'systemctl start ejabberd && systemctl start cron && systemctl start nginx && systemctl start nfs-kernel-server && /home/fabi/seafile-restart.sh && systemctl start netatalk && systemctl start monit'.
20190202-032634: MSG              02-02-2019 03:26:34 --- RBK0007I: Services werden gestartet: 'systemctl start ejabberd && systemctl start cron && systemctl start nginx && systemctl start nfs-kernel-server && /home/fabi/seafile-restart.sh && systemctl start netatalk && systemctl start monit'.
20190202-032634: DBG              -- systemctl start ejabberd && systemctl start cron && systemctl start nginx && systemctl start nfs-kernel-server && /home/fabi/seafile-restart.sh && systemctl start netatalk && systemctl start monit
20190202-032634: DBG                 >> executeShellCommand: systemctl start ejabberd && systemctl start cron && systemctl start nginx && systemctl start nfs-kernel-server && /home/fabi/seafile-restart.sh && systemctl start netatalk && systemctl start monit
20190202-032701: DBG                 << executeShellCommand: 0
20190202-032701: DBG              << startServices
20190202-032701: DBG              >> cleanupBackup
20190202-032701: DBG              -- Got trap EXIT
20190202-032701: DBG              -- rc: 109
Invocation parms: ''
20190202-032701: DBG                 >> sendEMail
20190202-032701: DBG                 -- Appendlog -a /var/log/raspiBackup/Raspbian.log
20190202-032701: DBG                 -- Sending eMail with program ssmtp and parms ''
20190202-032701: DBG                 -- Parm1:??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen. Parm2:Raspbian: Backup nicht erfolgreich !!!.
20190202-032701: DBG                 -- Content: 
02-02-2019 03:00:02 --- RBK0009I: Raspbian: raspiBackup.sh V0.6.4.2 (2a842c9) Sat Feb  2 03:00:02 CET 2019 gestartet.
02-02-2019 03:00:05 --- RBK0128I: Logdatei ist /var/log/raspiBackup/Raspbian.log.
02-02-2019 03:00:06 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
02-02-2019 03:00:06 --- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
02-02-2019 03:00:14 --- RBK0151I: Backuppfad /media/Server-Backup wird benutzt.
02-02-2019 03:00:14 --- RBK0008I: Services werden gestoppt: 'systemctl stop monit && systemctl stop nfs-kernel-server && systemctl stop cron && systemctl stop nginx && /home/fabi/seafile-stop.sh && systemctl stop netatalk && systemctl stop ejabberd'.
02-02-2019 03:00:33 --- RBK0081I: Backup vom Typ tar wird in /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190202-030001 erstellt.
02-02-2019 03:00:34 --- RBK0036I: Partitionslayout wird gesichert.
02-02-2019 03:00:34 --- RBK0044I: Backup der Bootpartition wird in /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190202-030001/Raspbian-backup.img erstellt.
02-02-2019 03:01:08 --- RBK0044I: Backup des Partitionlayouts wird in /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190202-030001/Raspbian-backup.sfdisk erstellt.
02-02-2019 03:01:08 --- RBK0046I: Backup des Masterbootrecords wird in /media/Server-Backup/Raspbian/Raspbian-tar-backup-20190202-030001/Raspbian-backup.mbr erstellt.
02-02-2019 03:01:08 --- RBK0158I: tar Backup "/media/Server-Backup/Raspbian/Raspbian-tar-backup-20190202-030001/Raspbian-tar-backup-20190202-030001.tar" wird erstellt.
02-02-2019 03:01:08 --- RBK0085I: Backuperstellung vom Typ tar gestartet. Bitte Geduld.
02-02-2019 03:26:24 ??? RBK0021E: Backupprogramm des Typs tar beendete sich mit RC 1.
[... cut ...]
20190202-032704: DBG              -- Removing mailfile /tmp/raspiBackup.maillog
20190202-032704: DBG              << cleanupTempFiles
20190202-032704: DBG           -- Terminate now with rc 109
02-02-2019 03:27:04 --- RBK0010I: Raspbian: raspiBackup.sh V0.6.4.2 (2a842c9) Sat Feb  2 03:27:04 CET 2019 beendet.
20190202-032704: MSG           02-02-2019 03:27:04 --- RBK0010I: Raspbian: raspiBackup.sh V0.6.4.2 (2a842c9) Sat Feb  2 03:27:04 CET 2019 beendet.</pre></p></details>

is this because of the same append_log failure with mpack not installed?

framps, 2019-02-02:
> Isn' it technically possible to just give a warning but still finishing the backup?

I agree. It's too harsh to abort the backup when the startup of a service fails. A message is already written I changed the code accordingly and in next version raspiBackup will not fail any more.

> /usr/local/bin/raspiBackup.sh: line 2387: mpack: command not found

Looks like you use msmtp. The check in the code whether mpack is installed misses a check for the new msmtp support. I added the check right now.

Both changes are commited in dev branch cfeb070e8139baa6e8d0e61b682f52cd5f6901a4

> But in the debug log the tar-Backup also fails with RC1

For RBK0021E see [this page](https://www.linux-tips-and-tricks.de/en/rmessages#m0021)

whowantsmybigdata, 2019-02-03:
> I agree. It's too harsh to abort the backup when the startup of a service fails. A message is already written I changed the code accordingly and in next version raspiBackup will not fail any more.

Cool thanks! 

>Looks like you use msmtp. The check in the code whether mpack is installed misses a check for the new msmtp support. I added the check right now.

Nice, but I use ssmtp if there is any difference...

→ after installing mpack (Debian package) and downloading the new raspiBackup script as in #108  all the sideproblems regarding the log and append-log functions are working perfectly. 
From night I finally got the email with detailled messages included and (debug) log attached as set in the config file without getting another message from cron.

Unfortunately I still get this RC1 tar-error for which I'm not even close to an idea where it is coming from. Although I set <code>DEFAULT_VERBOSE=1</code> I dont get any clues what might be wrong as it only says:
<pre> 
20190203-032354: DBG              << tarBackup 1
02-03-2019 03:23:54 ??? RBK0021E: Backupprogramm des Typs tar beendete sich mit RC 1.
20190203-032354: MSG           02-03-2019 03:23:54 ??? RBK0021E: Backupprogramm des Typs tar beendete sich mit RC 1.
20190203-032354: DBG              >> exitError 109
20190203-032354: DBG              << exitError 109
</pre> 
The link to the FAQ on your website you mentioned only says I should find the root cause or set <code> TAR_IGNORE_ERRORS="1" </code>

I found something in the log which might be interesting saying:
<pre>
-- Command executed:
tar 		-cpi --one-file-system 		 		 		-v 		-f "/media/Backup/Raspbian/Raspbian-tar-backup-20190203-030001/Raspbian-tar-backup-20190203-030001.tar" 		--warning=no-xdev 		--numeric-owner 		--exclude="/media/Backup/*" 		--exclude="//var/log/raspiBackup/Raspbian.log" 		--exclude='.gvfs' 		--exclude=/proc/* 		--exclude=/lost+found/* 		--exclude=/sys/* 		--exclude=/dev/* 		--exclude=/tmp/* 		--exclude=/boot/* 		--exclude=/run/* 		 		/
</pre>
Is it correct that it doesn't mention to exclude the mounted external drives  in <code>/media/*</code>? Or is this done by excluding <code>/dev/*</code>??

How can I find out the rootcause if there's no other verbose output?

framps, 2019-02-03:
> Nice, but I use ssmtp if there is any difference...

:-( will check again

> From night I finally got the email with detailled messages included and (debug) log attached as set in the config file without getting another message from cron.

Great. So we now have only one issue: the tar RC1 issue.

> The link to the FAQ on your website you mentioned only says I should find the root cause or set TAR_IGNORE_ERRORS="1" 

Don't do this. I have to delete this on my website. 

> Is it correct that it doesn't mention to exclude the mounted external drives in /media/? Or is this done by excluding /dev/??

No. --one-file-system makes sure no external mounted partition is saved

Quote from the [tar man page](http://man7.org/linux/man-pages/man1/tar.1.html)
```
If tar
              was given one of the --create, --append or --update options,
              this exit code means that some files were changed while being
              archived and so the resulting archive does not contain the
              exact copy of the file set.
```
This is an uggly error users of raspiBackup report this once a quarter. During backup some file or directory changed on your system and cause this error. I cannot help you on this. You have to find out which file/directory changed and stop the service modifying the file. 

Use option `--systemstatus` which creates a list of running services and open files just before the backup with tar starts in the debug log. This may give you an idea who's modifying a file. Option `-v` will also give you a detailed error message from tar. Check the debug log and inspect the -v output of tar carefully. All users of raspiBackup sooner or later found the file or directory which changed during backup. Either the service or open file list in the debug log helped or the tar error message.

In order to create the tar verbose output you also can invoke
`tar -cpi --one-file-system -v -f "/media/Backup/Raspbian/Raspbian-tar-backup-20190203-030001/Raspbian-tar-backup-20190203-030001.tar" --warning=no-xdev --numeric-owner --exclude="/media/Backup/*" --exclude="//var/log/raspiBackup/Raspbian.log" --exclude='.gvfs' --exclude=/proc/* --exclude=/lost+found/* --exclude=/sys/* --exclude=/dev/* --exclude=/tmp/* --exclude=/boot/* --exclude=/run/*  / 1>stdout.log 2>stderr.log` as root. Finally I hope you should find the tar error message only in stderr.log. If this doesn't work you have to check the whole tar log.

Crossing fingers you'll find the monkey changing a file during backup.

whowantsmybigdata, 2019-02-03:
I tried to stop as many services listed as active by <code>--systemstatus</code> as possible - no change in having a failure. 

I mean, I did successfull backups and restores with tar over raspiBackup since a few years with more or less the exact same configuration and services without having to stop more systemservices than mentioned in the beginning. Only things I have running while the backup are:
<pre>
 [ + ]  atd
 [ + ]  dbus
 [ + ]  dhcpcd
 [ + ]  dphys-swapfile
 [ + ]  fail2ban
 [ + ]  fake-hwclock
 [ + ]  irqbalance
 [ + ]  kmod
 [ + ]  minissdpd
 [ + ]  networking
 [ + ]  procps
 [ + ]  raspi-config
 [ + ]  rc.local
 [ + ]  rpcbind
 [ + ]  rsyslog
 [ + ]  ssh
 [ + ]  triggerhappy
 [ + ]  udev
 [ + ]  urandom
</pre>
which aren't stoppable afaik...

Interestingly while running the tar-command manually (after stopping the same services as when starting raspiBackup) stderr.log is only showing:
<pre>
tar: Removing leading `/' from member names
tar: Removing leading `/' from hard link targets
tar: /home/seafile/ccnet/ccnet.sock: socket ignored
tar: /home/fabi/stdout.log: file changed as we read it
</pre>
and the tar-archive is created with the correct filesize... so as far as I understand the tar-command was successfull?! So why the heck is raspiBackup failing?! 

I first thought the tar is failing while processing <code>/run/</code> (so indeed because of a running programm?!) as the debug log always shows the failure after processing this directory (which contents are excluded anyway). But after manually triggering tar a few times with different options I figured out <code>/run</code> simply is the last directory beeing processed, so in fact the tar is working all the way through but still sending raspiBackup a failure?! 

I have no clue...

whowantsmybigdata, 2019-02-04:
I tried a rsync backup with the exact same configuration and it worked flawlessly!! Now I would like to know if this rsync-backup is inconsistent, otherwise the tar-failure makes even less sense, doesn't it?

framps, 2019-02-04:
Keep cool. tar RC1 is uggly and was reported already a lot of time. 

> tar: /home/fabi/stdout.log: file changed as we read it 
That's the message which points to the root cause of RC1. Actually that's a file I told you to write into. I should have told you to redirect to some path on an external partition. My fault.
Try again with a different path and check the output.

> DEFAULT_LOG_OUTPUT=1
That's a deprecated option which may cause RC1. Delete it from your config.

> I tried a rsync backup with the exact same configuration and it worked flawlessly!! 
I don't know how tar and rsync detect and recover updates during backup processing. Maybe it's a sequence issue the files are processed? 

If your next reply is still rude I close this issue.

whowantsmybigdata, 2019-02-04:
Misunderstanding, I'm very sorry. Maybe its because of writing in english... but anyhow, I absolutely didn't want to be rude! I was just feeling quite devastated in trying to get a backup working for almost 2 weeks now although it worked perfectly since years and absolutely did not want to push you or anybody else to help me. Other way around, I'm not so used writing in forums to get help, mostly getting the help from doing research myself but with this problem I was totally running out of ideas so I'm actually very, very thankful for your great help here!

Will try without <code>DEFAULT_LOG_OUTPUT=1</code>. Thanks for the advice!

Please correct me if I'm wrong but the problem with writing stdout to a log file which is included in the backup itself is negligible I guess as the manual-tar-command seemed to work without failure exitcode and the output you mentionend seems to be just kind of a warning from tar (same like ignoring the socket, as far as I read in other forums)... Also, when doing a backup via raspiBackup this warning/failure doesn't show up in the logs as the path for the log file (in my case <code>/var/log/raspiBackup/Raspbian.log</code> is already excluded by the script?!

framps, 2019-02-05:
> and the output you mentioned seems to be just kind of a warning from tar

Yes, the socket error is not producing RC1 - but a file change. It's an error which should be handled.

> Will try without `DEFAULT_LOG_OUTPUT=1`. 

**No**. Either delete this config line and use the default or use `DEFAULT_LOG_OUTPUT=2` where the log is written into the backup directory.

> Also, when doing a backup via raspiBackup this warning/failure doesn't show up in the logs as the path for the log file (in my case `/var/log/raspiBackup/Raspbian.log` is already excluded by the script?!

Yes, it's excluded but I had users who still has issues with RC1 because of this. I don't want to do a code review of the tar code to understand the internal processing logic. That's why I decided to remove this option in 0.6.4.2 and store the log all the time in the backup location. For backward compatibility reasons this option is still there.

whowantsmybigdata, 2019-02-06:
> Yes, the socket error is not producing RC1 - but a file change. It's an error which should be handled.

Okay, I understood the tar-warning about sockets is just telling about the fact, tar is ignoring those sockets?! Everything I found in the internet is just advising to either ignore the warning or (better) exclude the socket(s) in the tar-command. The socket tar is warning about is of the ccnet backend of seahub (the webserver of seafile). As I completely stop seafile and seahub prior to starting the backup, the socket is useless and is only waiting to be destroyed/overriden by seahub while it starts the next time. From my point of view its kind of a bug from seahub leaving that socket untouched while stopping but anyhow...  Maybe it would be useful to include something like <code>--exclude=*.socket</code> in raspiBackup, as it would be simple to include in the script and wouldn't affect users not having a programm which fails to delete the socket(s) but could help those who have?!

> Log-ouput-dir

I completely took out the line regarding the output-dir of the log-file from the config. Good to know its saving the log in the backup-dir by default now. I once changed to the /var/log-option because raspiBackup deleted the log files when written to the backup-dir in case of a failed backup... if that problem is fixed I think its even more practical to have the log right beside the backup!

Now about my situation: rsync and dd backup tested with same config - both working. Tar-backup with excluding the ccnet.socket and with default log-directory-settings (line taken out from config) still having RC1-failure... 

If there are any further ideas it would be great to hear and test them, otherwhise I think I'll give up and switch to dd+rsync-backup...

framps, 2019-02-06:
> I changed to the /var/log-option because with the backupdir-option (which I would like more) there was always only a log in /var/log/raspiBackup - same happend now after deleting the line - log is saved in /var/log/raspiBackup nevertheless...

You should check raspiBackup messages :wink: `--- RBK0026I: Logfile saved in /root/raspiBackup.log.` I spent a lot of effort to have raspiBackup to be verbose as possible with messages.

> Now about my situation: rsync and dd backup tested with same config - both working without any problem.

Frankly I would use rsync if you have backupspace formatted with ext3/4. That's the best backup type you can use. tar makes sense only if you have a windows filesystem on your backupspace or use smb. dd backup is even worse. But people using Windows like this backup type :open_mouth: 

> Maybe it would be useful to include something like `--exclude=*.socket` in raspiBackup, as it would be simple to include in the script and wouldn't affect users not having a programm which fails to delete the socket(s) but could help those who have?!

A backup program should not hide anything going wrong or is weak in the backup process from the user. A user can exclude all files he want by adding exclude statements in options.

I just started tar backup on my system running also seafile 3 times and every time I got another error message from tar - /var/log/kernel changed, /var/log/message changed as we read it, I don't remember the third one. But every time I got a message from tar which told me which file changed. 

I see following options for you (sequence is my preference for you):
1. use rsync on a ext3/4 filesystem
2. interate and add exclude statements for files which vanish or change during backup
3. ignore RC1 with tar with option

whowantsmybigdata, 2019-02-07:
> You should check raspiBackup messages 😉 `--- RBK0026I: Logfile saved in /root/raspiBackup.log.` I spent a lot of effort to have raspiBackup to be verbose as possible with messages.

Yeah, I know but still... 
<pre>02-07-2019 03:00:02 --- RBK0009I: Raspbian: raspiBackup.sh V0.6.4.2 (2a842c9) Thu Feb  7 03:00:02 CET 2019 gestartet.
02-07-2019 03:00:03 --- RBK0128I: Logdatei ist /var/log/raspiBackup/Raspbian.log.
02-07-2019 03:00:04 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt. </pre>

Afterwards I tested with including <code>DEFAULT_LOG_OUTPUT=2</code> which worked as expected <pre>
02-07-2019 11:56:38 --- RBK0009I: Raspbian: raspiBackup.sh V0.6.4.2 (2a842c9) Thu Feb  7 11:56:38 CET 2019 gestartet.
02-07-2019 11:56:39 --- RBK0128I: Logdatei ist /media/Backup/Raspbian/Raspbian-rsync-backup-20190207-115638/raspiBackup.log.
02-07-2019 11:56:39 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.</pre>

but as soon as I commented out that line again it was jumping back to /var/log
<pre>02-07-2019 11:57:17 --- RBK0009I: Raspbian: raspiBackup.sh V0.6.4.2 (2a842c9) Thu Feb  7 11:57:17 CET 2019 gestartet.
02-07-2019 11:57:18 --- RBK0128I: Logdatei ist /var/log/raspiBackup/Raspbian.log.
02-07-2019 11:57:18 --- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.</pre>

I'll just keep the line in the config...

> A backup program should not hide anything going wrong or is weak in the backup process from the user. A user can exclude all files he want by adding exclude statements in options.

Good point but I thought having sockets backed up is senseless anyway?! So what do you mean by "going wrong or beeing weak"? Is my backup not constistent without sockets?

Anyhow, I will stick to your advice doing rsync only. Probably it would be key to have a spare microSD to test the restore (as you mention few times on your website anyway)...

Thanks a lot for your help and patience - I really appreciate your work (did you look into Bitcoins e.g. via Electrum already?)!!

framps, 2019-02-07:
> but as soon as I commented out that line again it was jumping back to /var/log

I checked the code and the default was 1. That's a bug I fixed right now in latest v0.6.4.2. 

> Good point but I thought having sockets backed up is senseless anyway?!

I agree. But this will require intense tests which are time consuming. Everybody is free to test the exclude statement very thoroughly. This is time consuming and even 1-2 SD cards may be damaged when executing tests. (You have to test all 3 backup types and this for all backup modes). 

> Anyhow, I will stick to your advice doing rsync only.

:+1:

> did you look into Bitcoins e.g. via Electrum already?

Yes, but I didn't create a wallet. Doesn't make sense for me.

Please close the issue when you're done.

whowantsmybigdata, 2019-02-08:
> That's a bug I fixed right now in latest v0.6.4.2

Good to know! 
Unfortunately I got the following output when upgrading to the newest version from your website and testing without <code>DEFAULT_LOG_OUTPUT=2</code>:
<pre> 
--- RBK0009I: Raspbian: raspiBackup.sh V0.6.4.2 (9172cd1) Fri Feb  8 12:02:43 CET 2019 gestartet.
??? RBK0027E: Kein externes Gerät an /backup verbunden. Die SD Karte würde für das Backup benutzt werden.
--- RBK0032I: Bitte warten bis aufgeräumt wurde.
^X--- RBK0105I: Neues Backupverzeichnis /backup/Raspbian/Raspbian-dd-backup-20190208-120242 wird gelöscht.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0010I: Raspbian: raspiBackup.sh V0.6.4.2 (9172cd1) Fri Feb  8 12:02:52 CET 2019 beendet.</pre>
Seems like the script is now completely ignoring the conf-file under <code>/usr/local/etc/raspiBackup.conf</code> as I actually have <code>DEFAULT_BACKUPPATH="/media/Backup"
</code> and <code>DEFAULT_BACKUPTYPE="rsync"</code>...

> I agree. But this will require intense tests which are time consuming.

True... there is much work behind opensource-developments - I'm sure most people (me mostly included) don't really think about that!

> I didn't create a wallet. Doesn't make sense for me.

Hm, okay up to you. I think it only takes a few minutes to set up securely and I guess there would be some donors beeing happy having an alternative besides Paypal...

framps, 2019-02-08:
> Seems like the script is now completely ignoring the conf-file under `/usr/local/etc/raspiBackup.conf`

Right. I fixed it right now. Just one line of code misplaced is sufficient to screw up the config handling :cry:  

> I'm sure most people (me mostly included) don't really think about that!

Most boring are tests which cannot be automated and thus have to be executed manually.  

>  I guess there would be some donors beeing happy having an alternative besides Paypal.

You can use Paypal just once to xfer some donations and delete the account again. Registration will take a few minutes :smile:

framps, 2019-02-09:
@whowantsmybigdata I just managed to create a [wallet](https://www.linux-tips-and-tricks.de/en/backup#donation). Looks like I used some wrong setup code the first time. As you said: It only takes a few seconds to setup a wallet :smiley:

whowantsmybigdata, 2019-02-09:
> Right. I fixed it right now. Just one line of code misplaced is sufficient to screw up the config handling 😢

Nice, tested it right now and it seems to work as expected...

>You can use Paypal just once to xfer some donations and delete the account again. Registration will take a few minutes

Haha, touché...

> I just managed to create a [wallet](https://www.linux-tips-and-tricks.de/en/backup#donation). Looks like I used some wrong setup code the first time. As you said: It only takes a few seconds to setup a wallet 😃

Great! I really guess this step will generate a few donations you might have missed otherwise... 
At least I will donate a little for great programm and great help here!

closed 2019-02-09 by whowantsmybigdata

framps, 2019-02-19:
@whowantsmybigdata Did you already donate some bitcoins? I didn't receive anything until now - or maybe there is still something missing in my electrum setup? Pls note: I'm not keen to get a donation but you asked me how you can donate bitcoins and now I want to practice bitcoin handling :smiley:

whowantsmybigdata, 2019-02-19:
Ja, hab ich am 14.2. an <code>13aXub22Sy3AvTKkj14yvnUdMQYJQFdbRo</code>. 
Steht auch als empfangen (12 Minuten später in die Blockchain geschrieben) da.
Kein Problem, kannst auch gerne nochmal fragen, falls es nicht klappt. 
Denke das hier kann auch auf Deutsch gehen, da es nicht mehr um das für andere evtl. relevante Problem geht, oder?

framps, 2019-02-20:
Hat geklappt :-) Mit der App sehe ich zwei Eingänge. Vielen Dank. Muss jetzt noch rausfinden warum der Linux Client Probleme hat.