framps/raspiBackup #118: Beta Installer - Update crontab time

coolduke666, 2019-03-15:
hi,

thx for the software... really good.

one problem that occur to me using the beta installer script: when you change the time of the cronjob, the script does NOT update the cronfile itself. you have to disable the cron and reenable it, in order to change the time for the cronjob.

please fix in the release... if there is one :)
thx

framps, 2019-03-15:
Thank you very much for the issue. I was able to reproduce and will fix it :smile:

coolduke666, 2019-03-15:
wow ... that was quick.

what a I also noticed in the beta installer: when switching vom "DD" with compression to rsync, you will get an error saying "cant use compression with rsync" when running a backup. I think the compression value will not be set to "0" when choosing "rsync"... 

when will the next version be released?

thx

framps, 2019-03-15:
Maybe you are interested to try the latest beta hot off the press version from [here](https://github.com/framps/raspiBackup/blob/ui_installer_beta/installation/raspiBackupInstallUI.sh).

framps, 2019-03-15:
Just verified your second issue. Codd catch. It's not fixed in the latest version. I'll fix it right now. Just watch for the next commit

framps, 2019-03-15:
Fixed with https://github.com/framps/raspiBackup/commit/f1ae7cdcd7ec0ce52e923fdd46bc402ad7021120

framps, 2019-03-15:
@coolduke666 Would you please verify both issues you reported are fixed and close this issue?

closed 2019-04-12 by framps