framps/raspiBackup #122: Add two new options which define commands executed before and after backup

framps, 2019-03-30:
Two new options should be available to execute custom commands just before starting the backup and just after finishing the backup.

framps, 2019-04-03:
@Hofei90 I just added your requested feature in the [development branch](https://github.com/framps/raspiBackup/blob/development/raspiBackup.sh)

Just add following two definitions in the config file:
```
# commands to execute before backup stops separated by &&
DEFAULT_BEFORE_STOPSERVICES='echo "Hello world I\'m stopping some stuff'"
# commands to execute after backup start separated by &&
DEFAULT_AFTER_STARTSERVICES='echo "Hello world I\'m starting some stuff'
```

Please verify the new feature. Don't use it in production because the dev branch is not ready for production.

framps, 2019-04-09:
Nein. Ich moechte im Installer nur die wichtigsten Dinge abhandeln. Ansonsten ist der Installer extrem aufwaendig zu programmieren und zu maintainen. UI Design und Programmierung ist einfach ziemlich aufwaendig. 
Dieses Feature zaehle ich zu den advanced Features und diese muss ein Benutzer nach wie vor in der Configdatei konfigurieren.

Hofei90, 2019-04-10:
So mein erstes Feedback dazu ist auch indirekt eine Frage.
Meine Config sieht so aus:

> # commands to execute before backup stops separated by &&
> DEFAULT_BEFORE_STOPSERVICES=python3 /home/pi/ausgabe_test.py >> /home/pi/test.log
> # commands to execute after backup start separated by &&
> DEFAULT_AFTER_STARTSERVICES=python3 /home/pi/ausgabe_test.py >> /home/pi/test.log

Das Skript ganz schlicht:
`print("Hello World")`

Der erhaltene Fehler lautet:
/usr/local/etc/raspiBackup.conf: Zeile 149: /home/pi/ausgabe_test.py: Keine Berechtigung

Dateistruktur:
> -rw-r--r--  1 pi   pi       21 Apr 10 14:24 ausgabe_test.py
> -rw-r--r--  1 root root      0 Apr 10 14:27 test.log

Ist es denn Absicht, dass die test.log als root erstellt wird, die Python Datei jedoch nur als User pi (nehme ich jetzt mal an) obwohl raspiBackup mit sudo gestartet worden ist?!
Irgendwie ist da ein Wurm drin?!

framps, 2019-04-10:
Please use either double quotes or quotes. Then your python program will be executed and the result file will be owned by root as you expect. 
```
# commands to execute before backup stops separated by &&
DEFAULT_BEFORE_STOPSERVICES='python3 /home/pi/ausgabe_test.py >> /home/pi/test.log'
# commands to execute after backup start separated by &&
DEFAULT_AFTER_STARTSERVICES='python3 /home/pi/ausgabe_test.py >> /home/pi/test.log'
```

Hofei90, 2019-04-11:
^^ Dann funktionierts auch - ich und bash/sh...

Habe jetzt im Pythonskript künstliche Verzögerungen bis zu 30 Sekunden eingebaut, Ausgaben kreirt, einzelne Befehle wie auch mit && Verknüpfte. Funktionierte alles wie erwartet.

Einziges noch zu klären, gäbs eine Timeoutzeit welche nicht überschritten werden darf, dann wär das noch zu testen.
Ansonsten würd ich sagen, es funktioniert wie gewünscht.

framps, 2019-04-11:
Klingt gut :-) 

Nein. Es gibt keinen Timeout. Alle Calls werden synchron ausgefuehrt.

Hofei90, 2019-04-11:
> Alle Calls werden synchron ausgefuehrt.
Ne && verkettete Befehle werden nacheinander abgearbeitet.
Oder wie meinst du das?

framps, 2019-04-12:
Da haben wir uns falsch verstanden. Die synchrone Reihenfolge der Ausfuehrungen ist wie folgt:

1. DEFAULT_BEFORE_STOPSERVICES 
2. DEFAULT_STOPSERVICES
3. Create Backup
4. DEFAULT_STARTSERVICES
5. DEFAULT_AFTER_STOPSERVICES 

Timeouts gibt es nicht in raspiBackup - ausser die ausgefuehrten Befehle haben irgendwelche timeouts.

Hofei90, 2019-04-13:
Passt, dann bin ich durch und sag es funktioniert

closed 2019-04-13 by framps