framps/raspiBackup #123: No services to stop causes error

normen, 2019-04-05:
OS: Raspbian jessie
raspiBackup V0.6.4.2

Description:
When no services are selected in the raspiBackupInstaller.sh and raspiBackup.sh is started it results in an error as seen below. I am running a read-only raspi and I really don't need to stop any services, now I have to stop a random service to make the script work :)

```
--- RBK0009I: makerpi: raspiBackup.sh V0.6.4.2 (74411c9) started at Fri  5 Apr 11:42:54 CEST 2019.
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0151I: Using backuppath /backup.
Too few arguments.
??? RBK0048E: Error occured when stopping services. RC 1.
--- RBK0033I: Please wait until cleanup has finished.
Too few arguments.
??? RBK0047E: Error occured when starting services. RC 1.
--- RBK0105I: Deleting new backup directory /backup/makerpi/makerpi-tar-backup-20190405-114254.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0010I: makerpi: raspiBackup.sh V0.6.4.2 (74411c9) stopped at Fri  5 Apr 11:42:56 CEST 2019.
```

To reproduce:
- Install raspiBackup using raspiBackupInstall
- Configure with no services selected
- Start backup
- Backup fails

Thanks for this awesome software! I have multiple raspis doing various things from controlling a 3d printer to controlling my lights and this script really has made backups SO easy! Keep up the awesome work!

normen, 2019-04-05:
I should note that I mean raspiBackupInstallUI beta when I say raspiBackupInstall!

framps, 2019-04-07:
Hello @normen ,

thank you very much for reporting this issue. Frankly there is already a new beta version available in the development branch which fixes this issue and improves the service selection menus but there were still some tests missing so I didn't publish it on my website until today. But now I published the version from the development branch on my website :smile:  

Would be great if you find some time to verify the current beta installer fixes you issue.

normen, 2019-04-08:
Yes indeed, that seems to fix it, thanks again for the great work!

framps, 2019-04-08:
Thank you very much for verifying. Would you please close the issue?

closed 2019-04-08 by normen