framps/raspiBackup #125: Installation raspiBackup im Development Branch mit UI

Hofei90, 2019-04-10:
Hallo,
wenn ich den UI installer im Developmentbranch starte, wird dennoch das stable raspiBackup installiert und nicht das vom Developmentbranch. Kann dies angepasst werden?

Ansich hab ichs schon hinbekommen, hab die sh Datei unter '/usr/local/bin/raspiBackup.sh' jetzt einfach manuell ersetzt. Hätte nur erwartet das man auch den Entwicklungsstand installiert bekommen würde.

framps, 2019-04-10:
Moin,

prinzipiell ginge das. Im Script sind die Downloadlocations Konstanten die man konfigurierbar machen muesste und die Du vermutlich angepasst hast.

Aber dann muesste ich mir eine Umgebung aufbauen die den Branch (master oder development) beruecksichtigt bzw das Deployment auf der Webseite. Ich spiele immer mal wieder mit dem Gedanken so eine Umgebung mit make aufzusetzen. Das ist aber ziemlich viel Arbeit, denn man muss alles auf einem Testrepo sorgfaeltig testen. Ansonsten kommt nur Schlechteres raus.

Hofei90, 2019-04-11:
> muesste und die Du vermutlich angepasst hast.
Nö, ich hab nur die raspiBackup.sh von git repo Verzeichnis in das  normale Installationsverzeichnis kopiert.
Ok, wenn das viel Arbeit ist...
War nur so ein Gedanke/eine Idee...

closed 2019-04-11 by Hofei90