framps/raspiBackup #128: Menge der Backups / Backup Strategie

DocYester, 2019-04-23:
Hallo,

wäre es möglich die aktuelle Backupstrategie (nur Menge an Tagen) ein wenig aufzubohren ?
So zB in die Richtung : https://opensource.com/article/18/8/automate-backups-raspberry-pi
Tage, Wochen, Monate, Jahre (Jahre müssen sicher nicht sein, aber sind eine Option)
Das Beispiel dort bezieht sich zwar nur auf rsync und würde sehr gut mit der Hardlink-Option arbeiten können, ist aber auch praktikabel für alle anderen Backup-Typen.

Danke

framps, 2019-04-23:
Die Idee ist gut. Ich denke ich habe auch eine Idee wie es konfiguriert und implementiert werden kann:

**Konfiguration:**
Bislang kann man bei `KEEPBACKUPS` nur eine Zahl angeben. Zukuenftig wuerde man noch mehr als eine Zahl angeben koennen, z.B. 
```KEEPBACKUPS="7 4 12 5" ```

D.h. die Semantik waere dann dass 7 Daily Backups vorgehalten werden. Wenn die Zahl 7 ueberschritten wird wird das aelteste Daily Backup ein neues Weekly Backup. Wenn die Zahl der Weekly Backups 4 ueberschreitet wuerde das aelteste Backup als Monthly Backup aufgehoben werden. Wenn dann die Zahl der Monthly Backups (12) ueberschritten wird wuerde das aelteste Backup als Yearly Backup vorgehalten. Wenn die Anzahl der Yearly Backups 5 uebersteigt wird das aelteste Backup geloescht.

So koennte man auch beliebige andere Strategien definieren:
```KEEPBACKUPS="7 4 3 4 5" ```
wuerde z.B. bedeuten, dass man 7 daily, 4 weekly, 3 monthly, 4 quartely und 5 yearly Backups vorhalten will. 

Die Backupverzeichnisse werden ihre Namen behalten anhand derer man feststellen kann, wann das Backup erstellt wurde. Es werden keine eyecatcher wie weekly, daily o.ae. eingefuegt.

Ist es das was Du Dir vorstellst?

DocYester, 2019-04-24:
Ja, das wäre genau das was ich meine.
Ob man dann alles einem Parameter konfiguriert, oder ob man 4 einzelne macht á la : 
KEEP_DAILY_BACKUPS=7
KEEP_WEEKLY_BACKUPS=3
..etc...
ist eigentlich Geschmackssache, wobei ich für 2teres wäre - ist verständlicher aufgrund der Namen :)
DIe Ordnernamen können alle so bleiben, wenn Du DIr das Beispiel auf der Seite ansiehst, ordnet das Script diese ja genau nach dem Prinzip.
Da die Namen der Ordner aktuell ja auch YYYY-MM-DD-HH-MM-SS aufgebaut sind, klappt das ohne Probleme.

framps, 2019-04-24:
Ich stimme Dir zu dass die Konfiguration einfacher zu verstehen ist mit dedizierten KonfigOptionsNamen. Aber dadurch verliert man die Möglichkeit eine andere Backupstrategie zu haben (z.B. 7 4 3 4 5). Ausserdem ist es in der naechsten Version 0.6.4.3 moeglich fuer jeden Backuptyp (dd, tar oder rsync) KEEP zu definieren. Wenn dann auch da noch dedizierte Optionsnamen benutzt werden sollen gibt es eine OptionsInflation. Deshalb bleibe ich bei meinem Ansatz. Ich denke mit zwei Beispielen kann man das Verhalten schon gut erklaeren.

Ich wuerde einen neuen Branch erstellen und das Feature da einbauen. Stellst Du Dich dann auch als Betatester zur Verfuegung bevor ich das Feature in die uebernaechste Version reinrolle? (Die naechste Version 0.6.4.3 ist schon Beta und wird dieses Feature nicht mehr bekommen).

DocYester, 2019-04-24:
Gerne, ich bin ohnehin gerade dran meine Raspi-Farm auf einen einheitlichen Stand zu bringen und auch die gesamte Backupstartegie zu überdenken und anzupassen.
Ich hätte also 1-2 Raspi's frei für Testings - kein Problem. 
Da das in der nächsten Version auch schon getrennt nach Backuptypen ist, sollten die Beispiele reichen, das stimmt.
War auch nur ein Gedanke meiner manchmal übertrieben gewünschten Übersichtlichkeit in Scripten und Config's :o)

framps, 2019-04-24:
:+1: Ich werde mich dann mal mit dem Thema befassen.

framps, 2019-04-25:
Ich habe mir die ganze Sache noch mal ueberlegt. Eigentlich ist es die Aufgabe von raspiBackup nur ein Backup zu erstellen. Eine BackupStrategie ist dann individuell unterschiedlich und sollte ausserhalb von raspiBackup definiert werden.

Dazu muss ich nur unendlich viele Backups zulassen und in dem Wrapperscript wird die Deletestrategie bzw der Code angewandt die Du verlinked hast.

Sprich, ich erweitere raspiBackup das man KEEP=0 angeben kann was das Löschen von Backups von raspiBackup ausschaltet und erweitere das Wrapperscript mit dem 7 4 12 5 BackupCleanupCode. Wer eine andere Backupstrategie haben moechte muss das Script anpassen. 

### Translated by framp

I relected my design.  raspiBackups role is to create backup - not to implement a backup strategy. That's something which has to be handled outside of raspiBackup.

So I just have to allow to disable deletion of backups and extent the wrapper script with the delete code you linked to.

Said that, I will extend raspiBackup to accept KEEP=0 to disable backup deletion and add the code in the wrapper script. Everybody who want's to use a different backup strategy can modify the wrapper script.

DocYester, 2019-04-25:
Stimmt, ist auch deutlich weniger Aufwand das verlinkte Script an's Ende des Backups einzubauen zum löschen und mit den Strategie-Parametern aus der Config zu füttern 👍

framps, 2019-04-28:
@DocYester I just created a [new branch](https://github.com/framps/raspiBackup/tree/backupStrategy) which includes a modified [raspiBackup](https://github.com/framps/raspiBackup/blob/backupStrategy/raspiBackup.sh) and a new wrapperscript [raspiBackup7424.sh](https://github.com/framps/raspiBackup/blob/backupStrategy/helper/raspiBackup7412.sh)

I tested the correct function of the backupStrategy by [creating dummy backup dirs and applying the backupStrategy](https://github.com/framps/raspiBackup/blob/backupStrategy/test/raspiBackup7412Fake.sh). Maybe this script can help you to perform your tests (I don't expect you to execute the test for the next 5 years :laughing: ).

Just let me know if you have any questions or findings when executing your test.

DocYester, 2019-04-30:
Tested the fake Wrapper - looks good so far - except one issue that i found : 
The monthly got a bug , one month was deleted that shouldnt ...
Execute backup strategy
--- Existing backups when backup strategy was applied ---
raspberry-rsync-backup-20150101-000000
raspberry-rsync-backup-20160101-000000
raspberry-rsync-backup-20170101-000000
raspberry-rsync-backup-20180101-000000
raspberry-rsync-backup-20180501-000000
raspberry-rsync-backup-20180601-000000
raspberry-rsync-backup-20180701-000000
raspberry-rsync-backup-20180801-000000
raspberry-rsync-backup-20180901-000000
raspberry-rsync-backup-20181001-000000
raspberry-rsync-backup-20181101-000000
raspberry-rsync-backup-20181201-000000
raspberry-rsync-backup-20190101-000000
raspberry-rsync-backup-20190301-000000
raspberry-rsync-backup-20190401-000000
raspberry-rsync-backup-20190408-000000
raspberry-rsync-backup-20190415-000000
raspberry-rsync-backup-20190422-000000
raspberry-rsync-backup-20190424-000000
raspberry-rsync-backup-20190425-000000
raspberry-rsync-backup-20190426-000000
raspberry-rsync-backup-20190427-000000
raspberry-rsync-backup-20190428-000000
raspberry-rsync-backup-20190429-000000
raspberry-rsync-backup-20190430-000000

1st Feb 2019 is missing ?!
After 3 runs, allways the same ...

framps, 2019-04-30:
Good catch :+1: I missed this bug in my tests :cry: .

I just used the logic from the website you linked to and thought the logic is OK :disappointed: . Actually I found out it's a [bug in the date command](http://bashworkz.com/linux-date-problem-1-month-ago-date-bug/)

I modified the code and added workaround code. Please try again :smiley:

DocYester, 2019-04-30:
I'm still testing around - it needs some more sorting logic and selecting the right backups (last ?!) from each day if there are mutltiple backups
i added some rnd and another loop here to test - will post it tomorrow with result from the next test without the date-bug 👍

framps, 2019-05-01:
Current code just picks the first backup of a day if there are multiple backups with `head -n 1`. `tail -n 1` will use the last one of a day. Do you think the last one should be selected? It's not a big deal to make this selection configurable. Question is what should be the default.

DocYester, 2019-05-01:
That's not completly correct ... it will pick the last/first file from the year ... time difference between is a year !
i made a pull request 👍

framps, 2019-05-04:
I merged your PR #131 and did some housekeeping in [this branch](https://github.com/framps/raspiBackup/tree/backupStrategy) If you close this issue I'll merge it into master.

DocYester, 2019-05-05:
👍

closed 2019-05-05 by DocYester

framps, 2019-05-05:
Just merged your PR in beta 0.6.4.3 branch.